OPCFoundation/UA-.NETStandard #1842: How to add user access restriction on nodes ?

GregoireG-C, 2022-06-08:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

I would to add access restriction on several server nodes as "SessionDiagnostics/ SessionSecurityDiagnostics" and "SessionDiagnosticsSummary" to make them unaccessible for anonymous users for instance. Can you give me some tips to do it ?
Thanks for your help !

### Environment

```markdown
- OS:Windows 10
- Environment:Visual Studio 2019
- Runtime:Net 4.6.2
```

mrsuciu, 2022-06-08:
@GregoireG-C 

Supposing you would like to restrict anonymous users to be able to access the node, than I would specify all the other user types which are allowed to access it using something in the lines of:

             `BaseVariableState sessionSecurityDiagnosticsArray = FindPredefinedNode(
                    VariableIds.Server_ServerDiagnostics_SessionsDiagnosticsSummary_SessionSecurityDiagnosticsArray,
                    typeof(BaseVariableState)) as BaseVariableState;
                sessionSecurityDiagnosticsArray.RolePermissions = new RolePermissionTypeCollection()
                {
                    // allow access to users except Anonymous role
                    new RolePermissionType()
                    {
                        RoleId = ObjectIds.WellKnownRole_SecurityAdmin,
                        Permissions = (uint)(PermissionType.Browse |PermissionType.Read|PermissionType.ReadRolePermissions | PermissionType.Write)
                    },
                    // new RolePermissionType() { ... }, and so on but excluding RoleId = ObjectIds.WellKnownRole_Anonymous
                };`

GregoireG-C, 2022-06-08:
Thanks a lot ! I will try this. I close the issue. If I found some difficulties, I will notice it.
See you :p

closed 2022-06-08 by GregoireG-C