OPCFoundation/UA-.NETStandard #1844: <title>

WipJesusFI, 2022-06-09:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

Hi!

I'm trying to connect a client (PC) to a server (PLC) using the UA-.NetStandard. I am using the ConsoleReferenceClient, adapting the serverUrl in the Program.Cs file and adding the username and the password required for the connection in the session creation in the UAClient.cs file. When I run the code I get the following error:

[image]

Any ideas where the problem is? I succesfully connected, with data exchange, via OPC UA the same PC with the same PLC using using both a sample client from https://integrationobjects.com/sioth-opc/sioth-opc-unified-architecture/opc-ua-client/ and a Python client programmed with the opcua library.

Thanks in advance

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

MD-V, 2022-06-09:
Hi @WipJesusFI,

if you are using ConsoleReferenceClient you should check UAClient.CertificateValidation https://github.com/OPCFoundation/UA-.NETStandard/blob/master/Applications/ConsoleReferenceClient/UAClient.cs#L274
for certificate validation logic. For your test I think you cold accept every certificate.

MD-V, 2022-06-09:
@WipJesusFI 

can you please check what the Error is? And if you are hitting the CertificateValidation?(https://github.com/OPCFoundation/UA-.NETStandard/blob/master/Applications/ConsoleReferenceClient/UAClient.cs#L284)

WipJesusFI, 2022-06-09:
@MD-V 

The error at that point of the CertificateValidation method is:

[image]

MD-V, 2022-06-09:
@WipJesusFI,

AutoAccept is overridden by:
https://github.com/OPCFoundation/UA-.NETStandard/blob/a7cfe2f2b70ead6e8c7eb1dd92ea3d56549ae1de/Applications/ConsoleReferenceClient/Program.cs#L164

Please make sure it is set correctly.

MD-V, 2022-06-09:
Yeah this is the expected behaviour.

WipJesusFI, 2022-06-09:
@MD-V 

thanks a lot :).

mregen, 2022-06-09:
There is also a `-a` command line option to set auto accept

closed 2022-06-10 by WipJesusFI