OPCFoundation/UA-.NETStandard #1847: Get server real name of structure instead of Opc.Ua.ExtensionObject type

WipJesusFI, 2022-06-10:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

Hi,

I have succesfully connected a PLC, which acts as a server, with a PC, which acts as a client, starting from the ConsoleReferenceClient project. I can read and write nodes from the PLC withouth any problem. 

Howewer, I am unable to read the red-boxed field shown below (the PLC-defined name of a complex type - structure).

[image]

Writing this code:

[image]

what I get is the following:

[image]

I wanna get a string or something similar containing **UDT_test_OPC** and not the type **Opc.Ua.ExtensionObject**.

Thanks in advance to the community

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

mregen, 2022-06-10:
Hi @WipJesusFI , these are complex types which are wrapped in a extension object. To decode, the typesstem must be loaded from the server. Here is a [sample code how to load it into a session.](https://github.com/OPCFoundation/UA-.NETStandard-Samples/blob/832eeb75732260cf14d3d049d3e4138529b602bb/Samples/NetCoreComplexClient/Program.cs#L374).
Once loaded, the client can decode the ExtensionObject and creates the objects with accessors that can be used to access the fields. Hope this helps...

WipJesusFI, 2022-06-13:
Thanks @mregen,

I did it with the following code:

[image]

getting the original complex type:

[image]

closed 2022-06-13 by WipJesusFI