OPCFoundation/UA-.NETStandard #1855: Cannot recreate the client session if server restarts because of different certificate data

backspacer80, 2022-06-24:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

After a restart of the server the serverCertificateData returned from the server is different with each restart of the server. For this reason the client's reconnect procedure will not work if the server becomes restarted during the client's lifetime.

I am using both the client and the server from this project. I'm working on stabilizing our system for cases where the client loses connection to the server. The client subscribes to the KeepAlive event and uses the SessionReconnectHandler to reconnect, as shown in the [ConsoleReferenceClient](https://github.com/OPCFoundation/UA-.NETStandard/blob/master/Applications/ConsoleReferenceClient/UAClient.cs). 

When the reconnection procedure kicks in then the client throws the following exception:

StatusCodes.BadCertificateInvalid, "Server did not return the certificate used to create the secure channel."

### Expected Behavior

The same server should return the same certificate raw data everytime it starts.
Moreover, if the client turns off security then the certificate data should not be compared at all in Session.Open.
The client should retain its session, including its subscriptions, after the server is restarted.

### Steps To Reproduce

1. Start the server.
2. Start the client and connect to the server.
3. Shutdown the server and restart it after a few seconds.
4. While the server is gone the client notices it in the KeepAlive handler and initiates the reconnect procedure.
5. The reconnect procedure uses SessionReconnectHandler which eventually calls Session.Recreate and Session.Open.
6. Session.Open compares the serverCertificateData returned from the server to m_endpoint.Description.ServerCertificate from the session that was used previously and fails because the server returns a different certificate after the restart.

A side note: I have security turned off in the client.

### Environment

```markdown
- OS: Windows 10 21H2
- Environment: Developer's notebook with Visual Studio
- Runtime: .NET 6
- Nuget Version: 1.4.368.58
- Component: Opc.Ua.Client, Session class
- Server: UA-.NETStandard-1.4.368.58
- Client: UA-.NETStandard-1.4.368.58
```

mregen, 2022-06-27:
> After a restart of the server the serverCertificateData returned from the server is different with each restart of the server. For this reason the client's reconnect procedure will not work if the server becomes restarted during the client's lifetime.

Sounds like your server creates a new app certificate everytime? Or is it just connecting with security None, where technically no cert is needed?
I need the info for repro.

backspacer80, 2022-06-27:
> Sounds like your server creates a new app certificate everytime? Or is it just connecting with security None, where technically no cert is needed?
Yes, we are currently running with security None.

mregen, 2022-08-23:
SessionReconnectHandler should detect the cert change and connect accordingly.

PR #2399 (2023-11-30): Implement async  reconnect
## Proposed changes

- Implement the reconnect and republish with async service calls. 

bug fixes:
- subscription Ids should start from a random number
- the keep alive interval calculation is too short, adding +1 for the timeout
- on BadSessionIdInvalid recreate session immediately 
- on a restarted server with new certificate, also reload endpoints when BadCertificateInvalid is returned

## Related Issues

- Fixes #2353
- Fixes #2320
- Fixes #1855 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can al [... cut]