OPCFoundation/UA-.NETStandard #1856: Certificate Validation - CRL certificates

vinodpydi, 2022-06-26:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

When i am running the CTT for Certificate Validation some test cases related to revocation list is failing and the error is ** Issued certificate of CA that has no revocation list available**? Am I missing something the configuration. Please help me at the earliest.

Thank you

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

mregen, 2022-06-27:
The CTT has a project folder `copyToServer`, from which you can copy the test certificates to the cert stores, also for the CRL. If the error occurs, CRL are probably not found for matching CA.

vinodpydi, 2022-06-27:
Thanks for your response. I have copied the required certificates to the respective folders in my certificate store. But still i get that the issue as no revocation list available. 
Any settings needs to be done in the config.xml file?

mregen, 2022-06-29:
The defaults in the ref server config are for the CTT testing, e.g. `<RejectUnknownRevocationStatus>true</RejectUnknownRevocationStatus>`, second check if you really copied to the right folders, naming is slightly different in ref server compared to the CTT `copyToServer` folder, e.g. `issuers` vs `issuer`.

mregen, 2022-08-23:
any updates, otherwise please close

vinodpydi, 2022-08-23:
Hi,

As suggested by you, followed the reference server and the folder structure. That solved my issue. Thank you !!!!!!!!!!!

closed 2022-08-23 by vinodpydi

vinodpydi, 2022-08-23:
Hi Team,

Thank you. I have closed the issue.

Regards,
Vinod Pydi