OPCFoundation/UA-.NETStandard #1858: Why does the standard OPCUA server sends its applicative certificate in PolicySecurity#None

GregoireG-C, 2022-06-28:
### Type of issue

- [ ] Bug
- [X] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

When an OPCUA client connects itself to the standard server, the server give its own applicative certificate into the EndpointDescription with the policy security #None into the GetEndpoint service response and into the answer of the CreateSession service.

In fact, this is compliant but I don't hunderstand this choice. The client does not need the applicative server certificate so why send it ?

> opcua part 4 - Services Release
> 
> GetEndpoints - §5.4.4.1 :
> "If the securityPolicyUri is NONE and none of the UserTokenPolicies requires encryption, the Client
> shall ignore the ApplicationInstanceCertificate."
> 
> CreateSession - §5.6.2.1:
> "Depending upon on the SecurityPolicy and the SecurityMode of the SecureChannel, the exchange 
> of ApplicationInstanceCertificates and Nonces may be optional and the signatures may be empty. 
> See Part 7 for the definition of SecurityPolicies and the handling of these parameters."
> 
> "f the securityPolicyUri is NONE and none of the UserTokenPolicies requires 
> encryption, the Client shall ignore the ApplicationInstanceCertificate."

An other thing, in the createSession method of StandardServer. there is a flag "requireEncryption " which is alway setted to true if the server can accept a secure identity token so I don't hunderstand its use since it's on the ActivateSession that the idendity check is made.

### Expected Behavior

The server does not send its own certificate in security policy #None

### Steps To Reproduce

1. Compile Standard UA server
2. Connect UAExpert with SecurityPolicy #None
3. You should have a pop up appear in UAExpert which ask you to accept the server certificate.

### Environment

```markdown
- OS: Windows 10 21H1
- Environment:Visual Studio 2019
- Runtime:Net 4.6.2
- Nuget Version:
- Component:
- Server: Reference Server
- Client:  UA Expert
```

### Anything else?

[image]

mregen, 2022-06-28:
The behavior has been changed a while back to return certificate only *not* if the endpoint is security None/anonymous user. Otherwise the cert is needed to send encrypted user token over the wire, or password would be sent as clear text.
see https://github.com/OPCFoundation/UA-.NETStandard/commit/8ddda78affed2b7344f50099661ae75c133a7aa8

mregen, 2022-07-07:
please close if answered...