OPCFoundation/UA-.NETStandard #1862: [PubSub] Publishing Events

stevicastojkovicabb, 2022-07-01:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

The ConsoleReferencePublisher has no example for publishing events but it turns out that there is no implementation for it.

Quote from: [ConsoleReferencePublisher/README.MD]( https://github.com/OPCFoundation/UA-.NETStandard/tree/master/Applications/ConsoleReferencePublisher)

```
Note:
The current PubSub implementation only supports PublishedDataItemsDataType as DataSetSource of a PublishedDataSetDataType from the configuration. Events will be added in a future version.
```

Any plans for it?

Thanks,
Stevica

### Environment

```markdown
- Nuget Version: 1.5.368.5-beta
- Component: Opc.Ua.PubSub
```

mrsuciu, 2022-08-11:
@stevicastojkovicabb There are no short term plans to support event publishing that I am aware of so far.

algorni, 2023-11-03:
does it means that the pubsub implementation here doesn't work?

closed 2025-12-03 by romanett