OPCFoundation/UA-.NETStandard #1864: Client based nodeset export

vinaybr, 2022-07-07:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

Using the OPC.Ua.Client library is it possible to connect to a server and export the address space to nodeset file? I am aware of the NodeStateCollection class which can be used on the server but i am intrested in a client based solution.

vinaybr, 2022-07-26:
@mregen can you please help me with this query

mregen, 2022-12-05:
This is currently investigated for a next release...
The console client sample already contains code to browse and import a complete address space of a server.

closed 2025-12-09 by marcschier