OPCFoundation/UA-.NETStandard #1865: <title>

vinaybr, 2022-07-07:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

Does the SDK have a way to validate if a node matches its type definition? 
For Ex - assume there is a type called MotorType which has 2 child variables. When i create a node of type BaseObjectType and set the HasTypeDefinition MotorType but dont create the child variable nodes or only create one of them or create them incorrect data types etc. 

Is there a way to enforce this validation.

vinaybr, 2022-07-26:
@mregen can you please help me with this query

EthanChangAED, 2023-07-21:
@mregen seems it's there is an easy answer for this. Can you help with answering?

mregen, 2023-07-21:
Hi @vinaybr, there is no helper function to do the validation. Is this needed for server or client side?

EthanChangAED, 2023-07-25:
Closing this issue. @vinaybr feel free to re-activate if you can provide us more background/justification about the necessity of built-in validation function.

closed 2023-07-25 by EthanChangAED