OPCFoundation/UA-.NETStandard #1873: SDK is not hard stop if the given application (Signed By CA) certificate is Invalid (Valid date expired)

utthungagokul, 2022-07-21:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [X] Help wanted

### Current Behavior

 
The current behavior is SDK still accepts the invalid CA signed certificate (Invalid expire date) and keep running with the Log where logged in  SDK logs and mentioned the SDK logs below for the reference.

************************* Logging started at 07/21/2022 14:20:54
07/21/2022 14:20:54.228 Imported the private key for [9B3CFE0CF7FE080E54382547D0C6139986071A13].
07/21/2022 14:20:54.804 Checking application instance certificate.
07/21/2022 14:20:54.859 Imported the private key for [9B3CFE0CF7FE080E54382547D0C6139986071A13].
07/21/2022 14:20:54.865 Check certificate: [CN=test, O=Org, L=Location, S=State, C=Country] [9B3CFE0CF7FE080E54382547D0C6139986071A13]
07/21/2022 14:20:54.872 Check application instance certificate. [CN=test, O=Org, L=Location, S=State, C=Country] [9B3CFE0CF7FE080E54382547D0C6139986071A13]
07/21/2022 14:20:55.029 Certificate Validation failed. Reason=BadCertificateTimeInvalid. [CN=test, O=Org, L=Location, S=State, C=Country] [9B3CFE0CF7FE080E54382547D0C6139986071A13]
07/21/2022 14:20:55.037 Application Certificate Validation suppressed BadCertificateTimeInvalid
07/21/2022 14:20:55.044 Certificate rejected. Reason=BadCertificateTimeInvalid. [CN=test, O=Org, L=Location, S=State, C=Country] [9B3CFE0CF7FE080E54382547D0C6139986071A13]
07/21/2022 14:20:55.093 Error validating certificate. Exception: Certificate has expired or is not yet valid. NotTimeValid: A required certificate is not within its validity period when verifying against the current system clock or the timestamp in the signed file.. Use certificate anyway?

 

### Expected Behavior

 Where I have the application which is using the SDK with application certificate which is signed by the Certificate Authority and had an usecase if the application certificate has expired date and deployed the expectation is SDK has to hard stop with exception.

**[Note: I'm running the client with console based application which will be running as services in the system]**

### Steps To Reproduce

1. Create the simple client code 
2. Create the CA singed certificate with invalid date
3. Run the client code 
4. Before calling the Create Session check the return value of "applicationInstance.CheckApplicationInstanceCertificate"

### Environment

```markdown
- OS: Windows 10
- Environment: Visual Studio 2019
- Runtime: .NET Core 3.1
- Nuget Version: 1.04.368
- Component: Opc.Ua.Configuration
- Server: 
- Client:
```

mregen, 2022-07-23:
Hi @utthungagokul , this behavior is by design to allow the server to start even when the certificate expired.

utthungagokul, 2022-07-26:
Hi @mregen, thanks for the replay. 

 I would like get the answer for the following question

1. Is there a recommendation for certificate validity duration?
2. Is the behavior of certificate validity enforceable using some configuration (block client/server if the certificate has expired)

mregen, 2022-07-28:
Hi @utthungagokul , 
1. you can check the [CA/Browser](https://cabforum.org/baseline-requirements-documents/) industry forum for guidance to run a PKI. At this time I'd not issue RSA app certs with a lifetime > 2 years. Some recommendations are even lower.
2. There are multiple ways to configure the certificate validator:
i ) config: set `ValidationOptions` on the certificate stores. e.g. `CertificateValidationOptions.SuppressCertificateExpired` = 1. e.g.
```xml
    <TrustedIssuerCertificates>
      <StoreType>Directory</StoreType>
      <StorePath>%LocalApplicationData%/OPC Foundation/pki/issuer</StorePath>
      <ValidationOptions>1</ValidationOptions>
    </TrustedIssuerCertificates>
```
ii) implementa a cert validation `CertificateValidationEventHandler` to accept the expired status codes.

mregen, 2022-08-17:
please reopen if not answered

closed 2022-08-17 by mregen