OPCFoundation/UA-.NETStandard #1874: ComplexType - Lack of Documentation

deinok, 2022-07-21:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

Documentation, Examples, etc... are inexistent regarding ComplexType

### Expected Behavior

Documentation, simple examples exists for the ComplexType usage.

I want to be able do Write and Read Structs from a Client to a Server in a one shot way.
I'm unable to figure out the way to do this.

mregen, 2022-07-23:
Hi @deinok, complex type bug fixes and updates are planned for the next release.

deinok, 2022-10-04:
Sorry, probably I think about Complex Types as a diferent thing.

How can I do something like:

```
public class MyComplexType {

    public string Variable1 {get; set;}

    public int Variable2 {get;set;}

}

var node = new NodeId(...)
var aMyComplexType = session.ReadComplexType<MyComplexType>(nodeId);
```

PR #1915 (2022-10-04): Complex types improvements
## Proposed changes

Improvements for complex types.
- Decode structures with multidimensional arrays
- Perf improvements for type loader using new NodeCache fetch multiple interfaces
- DataTypeDictionary validation when dependencies are used.
- interface for complex type resolver
- improvements of console client to demonstrate browse/read of custom values

## Related Issues

- Fixes #1874 
- fixes #1826 
- fixes #1715 
- fixes #920
- fixes #1848 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also f [... cut]