OPCFoundation/UA-.NETStandard #1879: GetIssuersNoExceptionsOnGetIssuer() may return false instead of true in some circumstances

DmitryFarberov, 2022-07-26:
There might be an issue with this method that maybe you could take a look at.

Say the call on line 676 in CertificateValidator.cs

**_(issuer, revocationStatus) = await GetIssuerNoException(certificate, m_issuerCertificateList, m_issuerCertificateStore, true).ConfigureAwait(false);_**

returns a valid issuer. In this case, shouldn't the isTrusted be set to true, just like it is in the case of the previous call to GetIssuerNoException() on line 665?

mregen, 2022-07-28:
Hi @DmitryFarberov , a certificate is only trusted if it is member of the trusted store, If it is member of the issuer store, it is only used to e.g. complete a cert chain, so the integrity of the certificate chain can be validated. Hope this makes sense..

mregen, 2022-08-08:
please reopen if not answered

closed 2022-08-08 by mregen