OPCFoundation/UA-.NETStandard #1880: CertificateIdentifier cannot find certificates in the LocalMachine store location

sammie-p, 2022-07-26:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

The CertificateIdentifier class cannot find or resolve the certificate given its SubjectName when the certificate is stored in the Windows LocalMachine store. I believe this is because internally when CertificateStoreIdentifier.CreateStore(StoreType) is called, it eventually calls new X509CertificateStore() which always sets the store location to CurrentUser. I can't find anywhere in the code path where the store location gets set to LocalMachine.

The problem does not exist when a certificate stored in the CurrentUser location is targeted instead of LocalMachine.

### Expected Behavior

It is expected that when setting the StorePath location to "LocalMachine\My" on an instance of CertificateIdentifier, that instance will enumerate the LocalMachine\My store location, and not the CurrentUser\My store location for the matching certificate when .Find() is called.

### Steps To Reproduce

`                
using Opc.Ua;
`
...
`
var astore = new CertificateIdentifier()
{
    StoreType = CertificateStoreType.X509Store,
    StorePath = "LocalMachine\\My",
    SubjectName = $"CN={Environment.MachineName}.domain" //a certificate well-known to exist
};
var cert = await astore.Find();

//cert == null
`

### Environment

```markdown
- OS: Windows 10
- Environment: Visual Studio 2022
- Runtime: .NET Core 6.0
- Nuget Version: 1.4.370.1
- Component: Opc.Ua.Core
```

ThomasNehring, 2023-06-06:
I'm not able to reproduce this. 

I just set up a reference server and a reference client to read a self-signed certificate from the MS certificate store. 

For this, I used the branch
https://github.com/OPCFoundation/UA-.NETStandard/tree/release/1.4.371 

The corresponding configuration entry in my config.xml files looks as follows:
    <ApplicationCertificate>
      <StoreType>X509Store</StoreType>
      <StorePath>LocalMachine\My</StorePath>
      <SubjectName>CN=Quickstart Reference Server, C=DE, S=BW, O=SAP SW, DC=XXXXXXXX</SubjectName>
    </ApplicationCertificate>

As you can see, I have replaced the generated certificate with a locally created one.

This works fine for both the server and the client, if you run the application as admin. Tested successfully with .Net 6.0 for the reference server, .Net48 for both reference client and server. When I connect to the server with a third party client it correctly shows the adapted subject, so the server did load that certificate from the MS store.

The certificate is read from within the CertificateIdentifier class. If you want to verify/debug this, put a break point in ApplicationConfiguration.cs, method
public virtual async Task Validate(ApplicationType applicationType)
in the line after the comment 

// load private key.

If you are still facing the issue please verify that your application is authorized to access the local machine store, and that you are using the CertificateIdentifier Find method in the way it is done in the Reference Server/Client.

ThomasNehring, 2023-06-06:
Since this cannot be reproduced resp it works fine in the reference server and client and the environment (.Net 6 and Windows 10) for which it was reported I now close this item.