OPCFoundation/UA-.NETStandard #1881: Subscription notification to include all monitored items if at least one item has changed

htbmw, 2022-07-27:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

Would it be possible to implement an opt-in feature to return all the monitored items belonging to a particular subscription when at least one of these monitored items has changed?

I had a look at the FastDataChangeCallBack of the subscription, and this is the closest I currently get to accessing the values of all monitored items that belong to the subscription.
The only problem for me is that the callback is provided with only the monitored items that have changed during the recent publishing cycle.

Is it possible / feasible to return all monitored items in the subscription when at least one of them has changed?

The idea behind this feature is to allow a client to group related monitored items into specific subscriptions, and then monitor each subscriptions as a whole, instead of handling the fine granular notifications provided by each monitored item's OnNotification event.

mregen, 2022-07-27:
Hi @htbmw, as I understand you look for a `ResendData` method that will only return all the last values of the current subscription. We just implemented the `ResendData` support on the server side because its mandatory for V1.04 compliance.
 Currently the only way to get the last value of all monitored items is to call the server method, but better were to have the client implementation. I heard request for such a feature many times (e.g. called heartbeat), so its about who can do it...

htbmw, 2022-07-29:
Hi @mregen 

I am not familiar with the ResendData specification. I will take a look at it when I get a chance.

I think you understand the idea of what I am looking for.
I am not only looking for availability of the feature itself, but also important for me is to understand if such a feature is feasible and would make sense in real world scenarios.
My biggest concern is whether such a feature will have an impact on the performance of either the server and/or client.

I have already done a quick proof of concept with a client implementation, and it is definitely possible to do this without any effect on the server's performance. 
My approach does not rely on the server to resend data, nor are there any read requests made to read the last values.

I will see if I can find the time to create a pull request and then this approach can be discussed further.
It does not necessarily have to be accepted, but should first serve as a topic for discussion and evaluation.

htbmw, 2022-08-04:
@mregen Would it be OK if I create a sample application in the samples repo to demonstrate my implementation?

I haven't found the best place yet where to put the enhanced feature in the shared libraries.

In my proof of concept, I just created a sub-class from Subscription, and implemented a callback for the FastDataChangeHandler that contains the logic to report all monitored items in the subscription when at least one item has changed.

I could perhaps also combine this with another proof of concept I did with setting up linked monitored items and using a triggering monitored item to trigger the reporting of the subscription.

I think this might be a great and valuable example for the community to see how you can use subscriptions in a more advanced  manner.

mregen, 2022-08-10:
Hi @htbmw , sounds great, please share your prototype in a branch of your github fork, I can check where the best location would be. one option is the samples repo but, or also integration in the `ÙaClient.cs` as a best practice might be an option.

EthanChangAED, 2024-03-29:
Close as won't fix for now. Feel free to reactivate if there are further updates, prototype or PR to be shared.

closed 2024-03-29 by EthanChangAED