OPCFoundation/UA-.NETStandard #1882: Can not found  "NodeManagerSaveFile" saved

sinfere, 2022-07-31:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [X] Help wanted

### Current Behavior

NodeManagerSaveFile path not exists after long running

### Expected Behavior

NodeManagerSaveFile exists

### Steps To Reproduce

using Quickstarts Servers ReferenceServer
1. add some nodes
2. run
3. wait

### Environment

```markdown
- OS: macOS
- Environment: JetBrains Rider
- Runtime: .NET Core 6.0
- Nuget Version:
- Component: Opc.Ua.Core
- Server: Reference Server
- Client:
```

EthanChangAED, 2023-07-21:
@sinfere are you still seeing this issue?

EthanChangAED, 2023-08-04:
Closing old bug that potentially fixed in new release.

closed 2023-08-04 by EthanChangAED