OPCFoundation/UA-.NETStandard #1883: ApplicationInstance.CheckApplicationInstanceCertificate throws exception for self signed certificate after Update from 1.4.367.42 to 1.4.369.30

tschombe, 2022-08-01:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

After update the nuget Package from Version 1.4.367.42 to 1.4.369.30 the follwing code, that worked before with self signed certifice throws an exception:

ApplicationInstance application = new ApplicationInstance();
application.ApplicationType = ApplicationType.Client;
var config = application.LoadApplicationConfiguration(@".\myOpcClient.Config.xml", false).Result;

//check the application certificate, if a new one is created, lifetime will be 240 month
application.CheckApplicationInstanceCertificate(true, 0, 240).Wait();

### Expected Behavior

The CheckApplicationInstanceCertificate() function validates the self created certificate

### Environment

```markdown
- OS: Windows 10
- Environment: Visual Studio 2022 17.2.6
- Runtime: .NET 4.8
- Nuget Version: 1.4.369.30
- Component: Opc.Ua.Configuration
- Server:
- Client:
```

MD-V, 2022-08-01:
@tschombe Can you please provide the Stack trace aswell?

tschombe, 2022-08-01:
Its a bit strange, first exception i receive are some C++ exceptions in the output window when i break when exception is thrown:
Exception thrown at 0x00007FFA4D48474C in Designer.exe: Microsoft C++ exception: EEFileLoadException at memory location 0x00000015256F5560.
The thread 0x4dd4 has exited with code 0 (0x0).
Exception thrown at 0x00007FFA4D48474C in Designer.exe: Microsoft C++ exception: [rethrow] at memory location 0x0000000000000000.
Exception thrown at 0x00007FFA4D48474C in Designer.exe: Microsoft C++ exception: EEFileLoadException at memory location 0x00000015256F5560.
Exception thrown at 0x00007FFA4D48474C in Designer.exe: Microsoft C++ exception: [rethrow] at memory location 0x0000000000000000.
Exception thrown at 0x00007FFA4D48474C in Designer.exe: Microsoft C++ exception: EEFileLoadException at memory location 0x00000015256F5560.
Exception thrown at 0x00007FFA4D48474C in Designer.exe: Microsoft C++ exception: Exception at memory location 0x00000015256F5F00.
Exception thrown at 0x00007FFA4D48474C in Designer.exe: Microsoft C++ exception: Exception at memory location 0x00000015256F5DE0.
Exception thrown at 0x00007FFA4D48474C in Designer.exe: Microsoft C++ exception: Exception at memory location 0x00000015256F5DE0.
Exception thrown at 0x00007FFA4D48474C in Designer.exe: Microsoft C++ exception: Exception at memory location 0x00000015256F6F40.
The thread 0x89c8 has exited with code 0 (0x0).
The thread 0x6e90 has exited with code 0 (0x0).
Exception thrown at 0x00007FFA4D48474C in Designer.exe: Microsoft C++ exception: [rethrow] at memory location 0x0000000000000000.
Exception thrown: 'System.IO.FileLoadException' in System.Formats.Asn1.dll
Die Datei oder Assembly "System.Runtime.CompilerServices.Unsafe, Version=4.0.4.1, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a" oder eine Abhängigkeit davon wurde nicht gefunden. Die gefundene Manifestdefinition der Assembly stimmt nicht mit dem Assemblyverweis überein. (Ausnahme von HRESULT: 0x80131040)

But this seems to be handled somewhere in the external code, but if i continue i get the following error:

Opc.Ua.ServiceResultException: 'Certificate validation failed. UntrustedRoot: Eine Zertifikatkette wurde zwar verarbeitet, endete jedoch mit einem Stammzertifikat, das beim Vertrauensanbieter nicht als vertrauenswürdig gilt.

Callstack:
   at Opc.Ua.CertificateValidator.<InternalValidate>d__35.MoveNext()

if i continue:
   at Opc.Ua.CertificateValidator.<InternalValidate>d__35.MoveNext()
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()

and if i continue further:
   at Opc.Ua.CertificateValidator.Validate(X509Certificate2Collection chain, ConfiguredEndpoint endpoint)

this exception was originally thrown at this call stack:
    Opc.Ua.CertificateValidator.InternalValidate(System.Security.Cryptography.X509Certificates.X509Certificate2Collection, Opc.Ua.ConfiguredEndpoint)
    System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
    System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(System.Threading.Tasks.Task)
    Opc.Ua.CertificateValidator.Validate(System.Security.Cryptography.X509Certificates.X509Certificate2Collection, Opc.Ua.ConfiguredEndpoint)

tschombe, 2022-08-01:
<?xml version="1.0" encoding="utf-8"?>
<ApplicationConfiguration
  xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
  xmlns:ua="http://opcfoundation.org/UA/2008/02/Types.xsd"
  xmlns="http://opcfoundation.org/UA/SDK/Configuration.xsd"
>
  <ApplicationName>OPC-UA Import 2</ApplicationName>
  <ApplicationUri>urn:localhost:opc_ua_import</ApplicationUri>
  <ProductUri>http://www.myuri.de/OpcImport</ProductUri>
  <ApplicationType>Client_1</ApplicationType>

  <SecurityConfiguration>
    
    <ApplicationCertificate>
      <StoreType>Directory</StoreType>
      <StorePath>CertificateStore\UA Applications\pki\own</StorePath>
      <SubjectName>OPC-UA Import 2</SubjectName>
    </ApplicationCertificate>

    
    <TrustedIssuerCertificates>
      <StoreType>Directory</StoreType>
      <StorePath>CertificateStore\UA Applications\pki\issuer</StorePath>
      <TrustedCertificates />
    </TrustedIssuerCertificates>

    
    <TrustedPeerCertificates>
      <StoreType>Directory</StoreType>
      <StorePath>CertificateStore\UA Applications\pki\trusted</StorePath>
      <TrustedCertificates />
    </TrustedPeerCertificates>

    
    <RejectedCertificateStore>
      <StoreType>Directory</StoreType>
      <StorePath>CertificateStore\UA Applications\pki\rejected</StorePath>
    </RejectedCertificateStore>

    
    <AutoAcceptUntrustedCertificates>true</AutoAcceptUntrustedCertificates>

    
    <RejectSHA1SignedCertificates>false</RejectSHA1SignedCertificates>
    <MinimumCertificateKeySize>2048</MinimumCertificateKeySize>

  </SecurityConfiguration>

  <TransportConfigurations>
    <TransportConfiguration>
      <UriScheme>http</UriScheme>
      <TypeName>Opc.Ua.Bindings.UaSoapXmlBinding</TypeName>
    </TransportConfiguration>
    <TransportConfiguration>
      <UriScheme>opc.tcp</UriScheme>
      <TypeName>Opc.Ua.Bindings.UaTcpBinding</TypeName>
    </TransportConfiguration>
  </TransportConfigurations>

  <TransportQuotas>
    <OperationTimeout>600000</OperationTimeout>
    <MaxStringLength>1048576</MaxStringLength>
    <MaxByteStringLength>1048576</MaxByteStringLength>
    <MaxArrayLength>65535</MaxArrayLength>
    <MaxMessageSize>4194304</MaxMessageSize>
    <MaxBufferSize>65535</MaxBufferSize>
    <ChannelLifetime>300000</ChannelLifetime>
    <SecurityTokenLifetime>3600000</SecurityTokenLifetime>
  </TransportQuotas>

  <ClientConfiguration>
    <DefaultSessionTimeout>60000</DefaultSessionTimeout>
    <WellKnownDiscoveryUrls>
      <ua:String>opc.tcp://{0}:4840</ua:String>
      <ua:String>http://{0}:52601/UADiscovery</ua:String>
      <ua:String>http://{0}/UADiscovery/Default.svc</ua:String>
    </WellKnownDiscoveryUrls>
    <DiscoveryServers xmlns:d3p1="http://opcfoundation.org/UA/2008/02/Types.xsd" />

    <MinSubscriptionLifetime>10000</MinSubscriptionLifetime>
  </ClientConfiguration>

  <Extensions>
    <ua:XmlElement>
      <ClientToolkitConfiguration>
        
        <DiscoveryOperationTimeout>10000</DiscoveryOperationTimeout>
        <DefaultKeepAliveInterval>5000</DefaultKeepAliveInterval>
        <SessionReconnectDelay>5000</SessionReconnectDelay>
        <DefaultSessionTimeout>60000</DefaultSessionTimeout>
        <DefaultSubscriptionPublishingInterval>1000</DefaultSubscriptionPublishingInterval>
        <DefaultSubscriptionKeepAliveCount>10</DefaultSubscriptionKeepAliveCount>
        <DefaultSubscriptionLifeTimeCount>1000</DefaultSubscriptionLifeTimeCount>
        <DefaultSubscriptionMaxNotificationsPerPublish>0</DefaultSubscriptionMaxNotificationsPerPublish>
        <DefaultSubscriptionPriority>255</DefaultSubscriptionPriority>
        <DefaultMiSamplingInterval>1000</DefaultMiSamplingInterval>
        <DefaultMiQueueSize>1</DefaultMiQueueSize>
        <DefaultEventMiQueueSize>0</DefaultEventMiQueueSize>
        <DecodeCustomDataTypes>true</DecodeCustomDataTypes>
        <DecodeDataTypeDictionaries>true</DecodeDataTypeDictionaries>
        <ClientCertificateLifeTime>240</ClientCertificateLifeTime>
      </ClientToolkitConfiguration>
    </ua:XmlElement>
  </Extensions>

  <TraceConfiguration>
    <OutputFilePath>..\Logs\OPCImport.log</OutputFilePath>
    <DeleteOnLoad>true</DeleteOnLoad>
	 
	
    
    
    
    
    
    
    
    
    
    
    
	  <TraceMasks>0</TraceMasks>
  </TraceConfiguration>

  <DisableHiResClock>true</DisableHiResClock>
</ApplicationConfiguration>

mregen, 2022-08-08:
@tschombe not sure whats going on here. Maybe clean up all Nuget dependencies, clean the build etc and start from scratch. Its unclear why you have a UaSoapXmlBinding in the text above, its not supported in this stack.

mregen, 2022-08-23:
any updates? I suspect a Nuget package mismatch.

tschombe, 2022-08-23:
Thanks for your reply, 
i think you are right, i tried to replicate the problem in a small sample application and everything works fine there.

It must be a Nuget mismatch with some assemblies messed up. Still investigating but i will close the item 
since it is a local issue and nothing global in the framework

closed 2022-08-23 by tschombe