OPCFoundation/UA-.NETStandard #1887: Snyk reports an issue with any project that references OPCFoundation.NetStandard.Opc.Ua

htbmw, 2022-08-04:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [X] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

We use Snyk for security scanning on a github repository with projects that reference the OPCFoundation stack.
Snyk reports an issue for each project that references the OPCFoundation.NetStandard.Opc.Ua package.

According to Snyk, the vulnerability is related to "Improper Certificate Validation":
https://security.snyk.io/vuln/SNYK-DOTNET-OPCFOUNDATIONNETSTANDARDOPCUA-1075446

You can see in the above link that it was reported on 17 February 2021, and that there is no fixed version for it yet.

More links related to the issue and provided by the Snyk scan report are:
https://www.cve.org/CVERecord?id=CVE-2020-29457
https://cwe.mitre.org/data/definitions/295.html

The vulnerability score assigned to the above issue is Medium: 6.5 
and is calculated according to the below calculator (link also provided by Snyk):
https://www.first.org/cvss/calculator/3.1#CVSS:3.1/AV:N/AC:L/PR:L/UI:N/S:U/C:H/I:N/A:N

Has this been reported here previously?
Is this something that can be fixed in the near future?

### Environment

```markdown
Current version of OPCFoundation packages being referenced:

1.4.368.58 

but I believe the version is irrelevant as there is no known fix available according to the report and links
```

ZbynekZ, 2022-08-05:
In the link you have provided yourself for more details, i.e., https://www.cve.org/CVERecord?id=CVE-2020-29457 , there is a link to https://opcfoundation.org/SecurityBulletins/OPC%20Foundation%20Security%20Bulletin%20CVE-2020-29457.pdf . That one implies that the issue is fixed in version 1.4.365.10.

Does that clarify the issue?

htbmw, 2022-08-05:
Yes that is perfect, thanks @ZbynekZ !
Somehow I missed that.

I am closing this issue.

closed 2022-08-05 by htbmw

mregen, 2022-08-05:
@htbmw I got to the same conclusion as @ZbynekZ. I opened a ticket at Snyk that their database is not quite right.

htbmw, 2022-08-05:
Thanks @mregen !
I was just now trying to figure out why their database is not up to date, but since you have already logged that ticket, then we can leave this.
Thank you!

htbmw, 2022-08-10:
@mregen Can you please confirm from your side whether the ticket you logged was completed?
I do not see any reports from Snyk, but it could be because I flagged it to be ignored.

mregen, 2022-08-10:
fyi 

> Ofir Eldar (Snyk) 
>Aug 7, 2022, 11:56 GMT+1 
>Hi there,
>
>We have released a fix to our vulnerability database that should be applied by the end of the day.
>
>Let us know if you do not see this change applied in the next 24 hours, or if you have any questions
>
>Have a great week!
>Kind regards,

htbmw, 2022-08-10:
Perfect, thanks @mregen 

I just now checked this link and it now correctly states how to fix it.
https://security.snyk.io/vuln/SNYK-DOTNET-OPCFOUNDATIONNETSTANDARDOPCUA-1075446

I think previously it stated "No known fix available", or something along this line.