OPCFoundation/UA-.NETStandard #1889: OPCFoundation.NetStandard.Opc.Ua Nuget package messes with Microsoft.Extensions.Hosting

odalet, 2022-08-04:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

Given this csproj:

```xml
<Project Sdk="Microsoft.NET.Sdk">
  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>net6.0</TargetFramework>
  </PropertyGroup>  
  <ItemGroup>
    <PackageReference Include="Microsoft.Extensions.Hosting" Version="6.0.1" />
    <!-- <PackageReference Include="OPCFoundation.NetStandard.Opc.Ua" Version="1.4.370.1" /> -->
  </ItemGroup>
</Project>
```

one has, alongside `Microsoft.Extensions.Hosting.dll`, a bunch of other; for example, `Microsoft.Extensions.Hosting.Abstractions.dll`:
[image]

If we uncomment `<PackageReference Include="OPCFoundation.NetStandard.Opc.Ua" Version="1.4.370.1" />` and rebuild, none of the dependencies of `Microsoft.Extensions.Hosting.dll` (in yellow in the screenshot above) are present anymore!

NB: Note that publishing the application changes nothing...

### Expected Behavior

Adding a reference to `OPCFoundation.NetStandard.Opc.Ua` should not prevent `Microsoft.Extensions.Hosting` dependencies from being copied over.

### Steps To Reproduce

1. Create a new .NET 6 console application with the csproj looking like the one above, 
2. Build
3. Examine the contents of the output folder

### Environment

```markdown
- OS: Windows 10
- Environment: Visual Studio 2022 17.2.2
- Runtime: .NET 6.0
- Nuget Version: 1.4.370.1
- Component:
- Server:
- Client:
```

mregen, 2022-08-05:
@odalet could you try first to add only `OPCFoundation.NetStandard.Opc.Ua.Core`?
The Opc.Ua umbrella packages pulls in also Https for backward compatibility. Maybe thats the root cause.

mregen, 2022-08-23:
any updates? other wise please close

odalet, 2022-08-23:
Hi sorry for the late reply, I was on something else... Anyway, I managed to get to the bottom of this.

**TLDR**: all of my problems came from the "ASP.NET Core 6 framework" being referenced in `Opc.Ua.Bindings.Https.csproj`:

```xml
<When Condition="'$(TargetFramework)' == 'netcoreapp3.1' OR '$(TargetFramework)' == 'net5.0' OR '$(TargetFramework)' == 'net6.0'">
  <ItemGroup>
    <FrameworkReference Include="Microsoft.AspNetCore.App" />
    <PackageReference Include="System.Net.Http" Version="4.3.4" />
  </ItemGroup>
</When>
```

**Details**: Originally my application was targeting .NET 4.8 and everything was fine. 

Then I wanted to move it to .NET 6 and I usually deploy these applications on machines where I only install the base .NET Core runtime.

However, because `Opc.Ua.Bindings.Https.csproj` directly reference the whole ASP.NET stack, I guess the compiler is clever and decides I don't need all these Microsoft.Extensions.* assemblies as they come with ASP.NET. The thing is, my target environment did not have ASP.NET and now neither did it have those dependencies...

I solved all of this by depending explicitly on "Microsoft.AspNetCore.App" in my codebase, the same way you do. Obviously, I also tweaked my deployment process so that it installs ASP.NET as well as the base .NET runtime.

I'm ok with having to do that, but this dependency is quite hidden and I wished it were explained somewhere in the readme: others may encounter the same issue.

mregen, 2022-08-24:
hi @odalet, if you don't expose a https endpoint then I'd recommend to just reference the server and config package.
For the ASP.NET framework reference, the problem is the kestrel server 2.x Nuget packages are not getting security fixes anymore, but thats the only way to get kestrel working with .NET framework. Starting with .NET Core 3.x the kestrel server comes as a framework bundle. The advantage is that all assemblies are wrapped in that framework reference and there is no more Nuget mismatch nightmare... sorry for any inconvenience this may cause. wrt ASP.NET. I *think* the Console samples don't need to reference a different framework unless they run in a docker container. Hope this helps.

closed 2022-08-30 by mregen