OPCFoundation/UA-.NETStandard #1893: Read Node Values of Custom DataType

matteomessmer, 2022-08-05:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

I am trying to read the value of this node: 
[image]

Usually I can read nodes of simple data type, but this is the first time I have to face this kind of data.

The datatype is this one:
[image]

Do you know how could I read it? I've read many forums and github posts about it, but so far I haven't found a working solution.

### Environment

```markdown
- OS: Windows 10
- Environment: Visual Studio 2022 17.1.6
- Runtime: .NET 6.0
- Nuget Version: v3 (?)
- Component: Opc.Ua, Opc.Ua.Client
- Server: Reference Server
- Client: UA Expert (Screenshots), My Test Client to test the read method
```

matteomessmer, 2022-08-11:
In the end I was able to read by adding to the node id the name of the field I want to read (i.e. ns=1;i=doG.Macch_Pronta_Marcia).

On the other hand I'm now facing the problem of writing on one of those fields. This method doesn't work, does someone have a solution?
Thank you,
Matteo

benjamin1988, 2022-08-11:
Hi @matteomessmer,

if you read the datavalue of the node, you propably will get an ExtensionObject. To Read or write this values you should rebuild the DataType in your .NET environment. May you can provide the datatype definition of the doG_typ out of the UA Expert. Then it would be possible to tell you have to build the class.

With kind Regards,
Benjamin

htbmw, 2022-08-20:
Take a look at the ComplexTypeSystem.
It builds .net types of complex types and abstracts the decoding of ExtensionObjects into strong typed objects

mregen, 2022-08-30:
Hi @matteomessmer , the PR #1915 shows samples how to read complex types. Also console client gets command line options to browse and read all complex types from a server and output as JSON.

mregen, 2022-12-11:
stale, please reopen if necessary.

closed 2022-12-11 by mregen