OPCFoundation/UA-.NETStandard #1905: BeginRead fails with NullReferenceException after reconnection is made using SessionReconnectHandler

jakub85, 2022-08-11:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [X] Help wanted

### Current Behavior

When using most recent released version of OPC UA stack v1.4.370.1 I'm getting NullReferenceException 

at Opc.Ua.SessionClient.BeginRead(RequestHeader requestHeader, Double maxAge, TimestampsToReturn timestampsToReturn, ReadValueIdCollection nodesToRead, AsyncCallback callback, Object asyncState) 
(this is the method at the top of the call stack provided by the raised Exception)

after the session is reconnected using SessionReconnectHandler, which is used when KeepAlive fails. Until reconnection happens, BeginRead operates normally and reads data from the server without problem.

Any idea how to handle the issue? Should I switch from APM to TAP - ReadAsync method, could it help? Or should I do the reconnection manually - close the session and reopen it without using SessionReconnectHandler (which requires me to keep track of subscribed points and resubscribe them, etc.)?

### Expected Behavior

No NullReferenceException and valid read result when session is reconnected using SessionReconnectHandler

### Steps To Reproduce

1. Create session with the OPC UA Server and monitor it using KeepAlive
2. Verify that BeginRead to any nodeId works fine
3. Simulate OPC UA Server failure so the KeepAlive detects a connection issue
4. As part of KeepAlive callback attempt to reconnect using SessionReconnectHandler
5. After the successful reconnection, re-attempt to read the same nodeId (as in step 2) using BeginRead

### Environment

```markdown
- OS: Windows 10
- Environment:
- Runtime: NET 6.0
- Nuget Version: 1.4.370.1
- Component:
- Server:
- Client:
```

mregen, 2022-08-12:
Hi @jakub85 , I suspect your BeginRead uses an old instance of session after reconnect, or the session was recoverd and the old session still works. Here is a pointer to a sample to use the session reconnecthandler complete: https://github.com/OPCFoundation/UA-.NETStandard/blob/885a61ba759e1328003089ce42691b9cc284cda9/Applications/ConsoleReferenceClient/UAClient.cs#L245

mregen, 2022-08-23:
any updates? otherwise please close.

mregen, 2022-08-30:
please reopen if necessary

closed 2022-08-30 by mregen