OPCFoundation/UA-.NETStandard #1912: <title>

mjaziri09, 2022-08-16:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

I have an OPC UA Server and I want to display the historical events from all nodes. From the server node I was able to monitor all the alarms in real time conditions. But when trying to display all the historical alarms , 0 values were returned.

### Expected Behavior

Display all the nodes historical alarms from the Server Node.

### Steps To Reproduce

1. Connect to an OPC UA Server simulator using an OPC UA Client (For me, I used uaExpert).
2. Add an event view document.
3. Drag and drop the Server node to the created document.
4. Choose Event History box and click on the refresh button to see the historical alarms.

### Environment

```markdown
- OS: Windows 10
- Environment:Visual Studio 2019 16.7.2
- Runtime: .NET Core 3.1
- Nuget Version: 1.4.368.33
- Component:Opc.Ua.Core
- Server:Integration Objects OPC UA Server Simulator
- Client: uaExpert
```

mregen, 2022-08-16:
Have you checked with the samples here: https://github.com/OPCFoundation/UA-.NETStandard-Samples/tree/master/Workshop/HistoricalEvents ? I don't know if the Integration Objects server supports historical events.

mjaziri09, 2022-10-11:
Hello Martin,

 

Thank you for your response. 

 

The Integration Objects Server support the historical alarms. Our need is to be able to monitor all the historical alarms available on all the nodes. It was possible to monitor all the server’s alarms by drag and drop the Server node to the Event Viewr document created in the uaExpert. But by doing the same to monitor the historical alarms an empty list was returned. It turns out that the Server node has 1 as the EventNotifier value, so there HistoricalRead access is denied. My question is : how to change the EventNotifier value of the Server node? 

 

 

Thanks & Regards,

closed 2023-06-16 by EthanChangAED