OPCFoundation/UA-.NETStandard #1913: "BadCertificateHostnameInvalid was returned during CreateSession" on first start of OPC server

tgaugler, 2022-08-16:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

Prerequisites:
- No Application certificate exists on server or client side - "first time use case"
- No User Security Policy:
```XML
<ServerSecurityPolicy>
        <SecurityMode>None_1</SecurityMode>
        <SecurityPolicyUri>http://opcfoundation.org/UA/SecurityPolicy#None</SecurityPolicyUri>
</ServerSecurityPolicy>
```
- Trust for all certificates
```XML 
<AutoAcceptUntrustedCertificates>true</AutoAcceptUntrustedCertificates>
``` 

Behavior:   
CheckApplicationInstanceCertificate creates new application certificate on first start. 
When a client tries to connect to the server with a URL that is not localhost/127.0.0.1 and not the DNS domain name, e.g. with the IP address of the OPC Server, the following code is executed, but on first start m_validatedCertificates is empty and the connect results in a "BadCertificateHostnameInvalid was returned during CreateSession" Error.
```C#
public void ValidateDomains(X509Certificate2 serverCertificate, ConfiguredEndpoint endpoint)
{
    X509Certificate2 certificate2;
    if (m_validatedCertificates.TryGetValue(serverCertificate.Thumbprint, out certificate2))
    {
        if (Utils.IsEqual(certificate2.RawData, serverCertificate.RawData))
        {
            return;
        }
    }
```

On first start validateDomains compares the IP address (used during connection) with the domain name specified in the auto generated certificate --> failure.

Error exists since 1.4.370.1.

It seems that the error got introduced in [StandardServer.cs](https://github.com/OPCFoundation/UA-.NETStandard/commit/274dfaddbae6118e23d4df74ea5ab3423586364e) with the introduction of the following code:
```C#
if (endpointUrl != null)
{
    // check the endpointurl
    ConfiguredEndpoint configuredEndpoint = new ConfiguredEndpoint() {
        EndpointUrl = new Uri(endpointUrl)
    };

    CertificateValidator.ValidateDomains(InstanceCertificate, configuredEndpoint);
}
```

As soon as the server is started a second time with existing certificate everything is ok. No error.

### Expected Behavior

Connect/createSession works on first start of server without error.

### Steps To Reproduce

1. Delete application certificates, e.g. under Linux from ~/.local/share/OPC Foundation and ~/.dotnet/corefx/cryptography/x509stores
2. Server Config 
```xml  
<AutoAcceptUntrustedCertificates>true</AutoAcceptUntrustedCertificates>
<ServerSecurityPolicy>
    <SecurityMode>None_1</SecurityMode>
    <SecurityPolicyUri>http://opcfoundation.org/UA/SecurityPolicy#None</SecurityPolicyUri>
</ServerSecurityPolicy>
```
2. Start Sample Server
4. Delete Certificates of UA Expert, e.g. from C:\Users\<username>\AppData\Roaming\unifiedautomation\uaexpert
5. Connect with UA Expert to server with IP address (not localhost/127.0.0.1/DNS name)
6. Custom Discovery, Connect with SecurityPolicy None
7. Trust server certificate
8. Connect to server

--> error  "BadCertificateHostnameInvalid was created during CreateSession"

On second start of OPC Server everything works perfectly, even if UA Experts certificates are deleted before. It is a problem on server side.

### Environment

```markdown
- OS: Debian 11
- Environment: VS Code
- Runtime: .Net 6
- Nuget Version: 1.4.370.1 - 1.4.370.12
- Component: Opc.Ua.Server
- Server: Reference Server
- Client: UA Expert
```

mregen, 2022-08-16:
Thanks for reporting @tgaugler, thats a bug. Apparently the domain check was added with 274dfaddbae6118e23d4df74ea5ab3423586364e to create a specific mismatch url audit event, but as far as I can tell the validator is a no op if the certificate is not new...

mregen, 2022-08-17:
Hi @tgaugler, the workaround for the special case here is to accept the status: `BadCertificateHostNameInvalid` in this event handler of your application, see sample: https://github.com/OPCFoundation/UA-.NETStandard/blob/0a9a3f3a7db845dcd5531168ebfd4d8de4fde89c/Applications/ConsoleReferenceServer/UAServer.cs#L213

PR #1914 (2022-08-30): Fix server url mismatch handling
## Proposed changes

- change the Certificate domain validation to work also for the server case
- do not refuse a connection if the check fails, only trigger the audit event

## Related Issues

- Fixes #1913 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds functionality)
- [ ] Test enhancement (non-breaking change to increase test coverage)
- [ ] [... cut]
closed 2022-08-30 by mregen