OPCFoundation/UA-.NETStandard #1921: CVE detected in version 1.4.370.12

odalet, 2022-08-25:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

It seems that PR #1878 is raising a security alert (CVE-2022-33916): 

[image]

Associated links:

* <https://ossindex.sonatype.org/component/pkg:nuget/OPCFoundation.NetStandard.Opc.Ua@1.4.370.12>
* <https://ossindex.sonatype.org/vulnerability/CVE-2022-33916?component-type=nuget&component-name=OPCFoundation.NetStandard.Opc.Ua>
* <https://nvd.nist.gov/vuln/detail/CVE-2022-33916>

### Expected Behavior

No security alert :)

### Environment

```markdown
- Nuget Version: 1.4.370.12
- Component: Opc.Ua.Server
```

odalet, 2022-08-25:
It is very weird:
* The CVE is not raised if I use v1.4.370.9
* It does if I use v1.4.368.58

How come 1.4.370.9 is not vulnerable if the version after is and a version before is as well...

I've also been reading <https://files.opcfoundation.org/SecurityBulletins/OPC%20Foundation%20Security%20Bulletin%20CVE-2022-33916.pdf> but it's really not clear to me whether:
* commit #313aa2a2499d8690cf719a67176e131517bb8b78 (aka PR #1878) addresses the issue (in which case, v1.4.370.12 should not be vulnerable)
* Or this commit introduces the issue, in which case v1.4.370.12 is vulnerable (and v1.4.370.9 is not), why is v1.4.368.58 vulnerable?

mregen, 2022-08-29:
Hi @odalet, the issue is fixed with any release after 1.4.370.1 (except 1.5.xxx prereleases), so this is a false alert.

odalet, 2022-08-29:
Thanks for the information!

closed 2022-08-29 by odalet