OPCFoundation/UA-.NETStandard #1928: Session.Read doesn't seems to timeout

utthungagokul, 2022-09-12:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [X] Help wanted

### Current Behavior

Hi Team,

  Currently I'm facing an issue with data loss , when the server are disconnecting from the network ,one of the read call is not getting error/response from the read call.

 In my scenario I have a timer with 3sec interval to read the  particular node  Id , when server goes down from the network one of the read call not return the error/response.

[image]

### Expected Behavior

When there is no connection Session.Read should return the error/response.

### Environment

```markdown
- OS:
- Environment:
- Runtime: 
- Nuget Version: 1.4.370.12
- Component:
- Server:
- Client:
```

mrsuciu, 2022-09-16:
As I understand you are using a synchronous read call.
You should at least receive an exception (ex: BadSecureChannelClosed) if the server is unresponsive.
Please share a snippet of code which reproduces the situation.

EthanChangAED, 2023-06-20:
Close and dup to https://github.com/OPCFoundation/UA-.NETStandard/issues/1091

closed 2023-06-20 by EthanChangAED