OPCFoundation/UA-.NETStandard #1933: Retrieving all alarm nodes from the server

MaciejChodorowski, 2022-09-16:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

Hello

My goal is to retrieve all alarm nodes (their addresses) from the server. These nodes are characterized by having _HasTypeDefinition_ reference pointing towards the ObjectType named **AlarmType**.

Our existing solution is very slow and most likely suboptimal. Currently it works by searching the whole tree and comparing each node with custom expression, in this case, checking if the nodes reference table contains reference named **AlarmType**.

I don't think this is the right direction to solve this problem and I doubt how much it is possible to speed up this proccess. My idea was to use the **AlarmType** reference and by reversing it reaching the desired nodes. Instead this reference pointed back not at these nodes, but to different SubTypes of **AlarmType**. These SubTypes have matching BrowseName/DisplayName to the target alarm nodes but are not connected in any way to them. Overall, this solution failed.

What would be the optimal way of achieving my goal?

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

jh-isw, 2022-09-18:
You could try to limit the load by specifiying the nodeClassMask and resultMask in the Browse.
[https://reference.opcfoundation.org/Core/Part4/5.8.2/#5.8.2.2](https://reference.opcfoundation.org/Core/Part4/5.8.2/#5.8.2.2)

mregen, 2022-12-11:
I *think* its sufficient to browse the server object with hierarchical reference `HasNotifier`.

EthanChangAED, 2023-07-27:
Converting this to Q&A and mark as answered.