OPCFoundation/UA-.NETStandard #1939: Application Certificate Issue in MAUI Project on M1 Macbook

yagizhanNY, 2022-09-28:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [X] Help wanted

### Current Behavior

I am trying to develop OPC UA client application in MAUI project on M1 Macbook.

I am facing some issues about application certificate.

You can find my config file at below.

```xml
<?xml version="1.0" encoding="utf-8"?>
<ApplicationConfiguration
  xmlns:ua="http://opcfoundation.org/UA/2008/02/Types.xsd"
  xmlns="http://opcfoundation.org/UA/SDK/Configuration.xsd"
  schemaLocation="./Schema/ApplicationConfiguration.xsd"
>
  <ApplicationName>UA Core Sample Client</ApplicationName>
  <ApplicationUri>urn:localhost:OPCFoundation:CoreSampleClient</ApplicationUri>
  <ProductUri>http://opcfoundation.org/UA/CoreSampleClient</ProductUri>
  <ApplicationType>Client_1</ApplicationType>

  <SecurityConfiguration>

    <!-- Where the application instance certificate is stored (MachineDefault) -->
    <ApplicationCertificate>
      <StoreType>Directory</StoreType>
      <StorePath>%LocalApplicationData%/OPC Foundation/pki/own</StorePath>
      <SubjectName>CN=YNY UA Client</SubjectName>
    </ApplicationCertificate>

    <!-- Where the issuer certificate are stored (certificate authorities) -->
    <TrustedIssuerCertificates>
      <StoreType>Directory</StoreType>
      <StorePath>%LocalApplicationData%/OPC Foundation/pki/issuer</StorePath>
    </TrustedIssuerCertificates>

    <!-- Where the trust list is stored (UA Applications) -->
    <TrustedPeerCertificates>
      <StoreType>Directory</StoreType>
      <StorePath>%LocalApplicationData%/OPC Foundation/pki/trusted</StorePath>
    </TrustedPeerCertificates>

    <!-- The directory used to store invalid certficates for later review by the administrator. -->
    <RejectedCertificateStore>
      <StoreType>Directory</StoreType>
      <StorePath>%LocalApplicationData%/OPC Foundation/pki/rejected</StorePath>
    </RejectedCertificateStore>

    <!-- WARNING: The following setting (to automatically accept untrusted certificates) should be used
    for easy debugging purposes ONLY and turned off for production deployments! -->
    <AutoAcceptUntrustedCertificates>false</AutoAcceptUntrustedCertificates>

  </SecurityConfiguration>

  <TransportConfigurations/>

  <TransportQuotas>
    <OperationTimeout>600000</OperationTimeout>
    <MaxStringLength>1048576</MaxStringLength>
    <MaxByteStringLength>4194304</MaxByteStringLength>
    <MaxArrayLength>65535</MaxArrayLength>
    <MaxMessageSize>4194304</MaxMessageSize>
    <MaxBufferSize>65535</MaxBufferSize>
[... 206 lines cut ...]

[image]

Also I tried to use X509Store for application certificate. At that time app gives the `System.Security.Cryptography.CryptographicException a required entitlement isn't exist` exception.

These codes are working well on Windows 10 machine.

### Expected Behavior

I want to store application certificate automatically when app is started in directory or store on M1 Macbook. 

### Environment

```markdown
- OS: MacOS Monterey 12.6 (Apple M1)
- Environment: Visual Studio for MAC 17.3.5, XCode 13.4.1
- Runtime: .Net 6.0.401
- Nuget Version: 6.2.1.2
- Component: Opc.Ua.Client
- Server: Opc UA Device
```

### Anything else?

Is there anybody faced this issue before? Do somebody know the solution?

LaXiS96, 2022-10-12:
Hello,
I have the same issue, my app was running fine in .NET Core 3.1 with v1.4.369.30, but then after upgrading to .NET 6 and v1.4.370.12, certificates are still getting created as expected but aren't getting loaded properly, failing with the same `There is no cert with subject ...` error.

After debugging for a while, it looks like the problem starts at [ApplicationInstance.cs#L808](https://github.com/OPCFoundation/UA-.NETStandard/blob/1.4.370.12/Libraries/Opc.Ua.Configuration/ApplicationInstance.cs#L808), where the certificate (and associated private key) that was just created is reloaded from disk.

Then we have [DirectoryCertificateStore.cs#L391](https://github.com/OPCFoundation/UA-.NETStandard/blob/1.4.370.12/Stack/Opc.Ua.Core/Security/Certificates/DirectoryCertificateStore.cs#L391) which always throws:
```plain
Internal.Cryptography.CryptoThrowHelper+WindowsCryptographicException: The system cannot find the path specified.
   at Internal.Cryptography.Pal.CertificatePal.FromBlobOrFile(ReadOnlySpan`1 rawData, String fileName, SafePasswordHandle password, X509KeyStorageFlags keyStorageFlags)
   at System.Security.Cryptography.X509Certificates.X509Certificate..ctor(String fileName, String password, X509KeyStorageFlags keyStorageFlags)
   at System.Security.Cryptography.X509Certificates.X509Certificate2..ctor(String fileName, String password, X509KeyStorageFlags keyStorageFlags)
   at Opc.Ua.DirectoryCertificateStore.LoadPrivateKey(String thumbprint, String subjectName, String password)
```

It looks like the issue is caused by the runtime (.NET) and not the OPC UA library itself... (in fact, this section of code hasn't been changed between 1.4.369.30 and 1.4.370.12)

yagizhanNY, 2022-10-16:
> Hello, I have the same issue, my app was running fine in .NET Core 3.1 with v1.4.369.30, but then after upgrading to .NET 6 and v1.4.370.12, certificates are still getting created as expected but aren't getting loaded properly, failing with the same `There is no cert with subject ...` error.
> 
> After debugging for a while, it looks like the problem starts at [ApplicationInstance.cs#L808](https://github.com/OPCFoundation/UA-.NETStandard/blob/1.4.370.12/Libraries/Opc.Ua.Configuration/ApplicationInstance.cs#L808), where the certificate (and associated private key) that was just created is reloaded from disk.
> 
> Following the call stack we then have [DirectoryCertificateStore.cs#L391](https://github.com/OPCFoundation/UA-.NETStandard/blob/1.4.370.12/Stack/Opc.Ua.Core/Security/Certificates/DirectoryCertificateStore.cs#L391) which always throws:
> 
> ```
> Internal.Cryptography.CryptoThrowHelper+WindowsCryptographicException: The system cannot find the path specified.
>    at Internal.Cryptography.Pal.CertificatePal.FromBlobOrFile(ReadOnlySpan`1 rawData, String fileName, SafePasswordHandle password, X509KeyStorageFlags keyStorageFlags)
>    at System.Security.Cryptography.X509Certificates.X509Certificate..ctor(String fileName, String password, X509KeyStorageFlags keyStorageFlags)
>    at System.Security.Cryptography.X509Certificates.X509Certificate2..ctor(String fileName, String password, X509KeyStorageFlags keyStorageFlags)
>    at Opc.Ua.DirectoryCertificateStore.LoadPrivateKey(String thumbprint, String subjectName, String password)
> ```
> 
> It looks like the issue is caused by the runtime (.NET) and not the OPC UA library itself... (in fact, this section of code hasn't been changed between 1.4.369.30 and 1.4.370.12)
> 
> I believe this issue might be related: [dotnet/runtime#25143](https://github.com/dotnet/runtime/issues/25143) Also pinging @niveditha-sooda as the author of the implementation.

Thanks for the answer. Did you find any workaround or is it still open issue at your side?

LaXiS96, 2022-10-17:
@yagizhanNY No workaround at the moment, because the issue seems to depend on the environment which the app is running in (for example, in the referenced runtime issue they talk about an missing user profile).
In fact, I am only noticing this behavior when integration testing via xUnit in VS2022 and not while debugging straight from VS2022 or running the published app.

LaXiS96, 2022-10-18:
I believe to have found the culprit, and indeed my tests are now working.
The cause was the length of the full path to the certificate/private key, which in my case was more than 255 characters and supposedly that's too many for the native API that `X509Certificate2` uses.
My `pkiRoot` base path was 167 characters long, to which you need to add `\private\<applicationName> [<fingerprint>].pfx` for a total of 261.
For the record, I am running on Windows 10 build 19041.1415.

yagizhanNY, 2022-10-18:
@LaXiS96 Very good news! But I tried with shorter path on my M1 Macbook `/Users/yagizhan/own/certs/Pre-Dev [C2B92DE8AC9FC30FAE99200FD912037AA535FD74].der`. It didn't worked. Did you change anything else?

LaXiS96, 2022-10-19:
In that case I believe it could be related to the fact that you are running on Mac, have you tried running your project (or the reference samples) on a Windows/Linux machine?
Anyway this issue is not directly caused by the OPC UA library but by the .NET runtime, so it might be better to look at existing issues or open a new one in [dotnet/runtime](https://github.com/dotnet/runtime).

yagizhanNY, 2022-10-27:
@mregen Yes, maybe you are right. It can be problem about M1 platform also my MacOS version is 12. I will keep debug it. If any updates, I will write here.

yagizhanNY, 2023-02-25:
I have an update. The issue still ongoing on .Net7. I created the application from scratch on my Macbook, then I tried to create the certificate in store, it worked. I am able to use certificate management with store but it still won't working with directory.

yagizhanNY, 2024-12-01:
Hey @emncnozge, could you check the latest status of the project and update this issue, please? Yusuf might know the context. Thanks!

closed 2025-12-03 by romanett