OPCFoundation/UA-.NETStandard #1942: Trusted issuer folder does not allow clients signed by trusted issuer to connect

elliot-gawthrop, 2022-10-06:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

When attempting to connect with a client whose certificate is signed by a trusted issuer, the server will reject the connection unless the trusted issuer is placed in the trusted peer store. Whether a trusted issuer is placed in the issuer store seems to make no difference. Its purpose is unclear.

### Expected Behavior

**Client configuration**
Client certificate: A
Issuer of A: B

**Server configuration**
B placed in neither trusted peer nor trusted issuer store -> connection rejected (as expected)
B placed into trusted issuer store -> connection rejected (why?)
B placed into trusted peer store -> connection accepted (why?)
B placed into trusted peer and trusted issuer store -> connection accepted

### Steps To Reproduce

1. Download the attached zip file
2. Extract the contents of the copyToClient directory to the clients pki root
3. Extract the contents of the copyToServer directory to the servers pki root
4. Configure the client to use the ReproClient own certificate
5. Start the server
6. Attempt to make a secure connection from the client

Expected: connection should be accepted
Actual: connection rejected

### Anything else?

[certificates.zip](https://github.com/OPCFoundation/UA-.NETStandard/files/9722774/certificates.zip)

Contains:
* copyToClient/own/certs/ReproClient.der
* copyToClient/own/private/ReproClient.pfx
* copyToServer/issuer/certs/ReproCA.der
* copyToServer/issuer/crl/ReproCA.crl

mregen, 2022-10-06:
Hi @elliot-gawthrop, the behavior you describe is exactly as expected.
However, the xml configuration entries and comments in samples are misleading.
The UA stack default security configuration follows this guideline: [Certificate Store Directory Layout](https://reference.opcfoundation.org/GDS/docs/F.1/). 
A trusted leaf or issuer certificate must be added to the store called `<TrustedPeerCertificates>`.
An issuer certificate that is just needed to complete a cert chain must be added to a store called `<TrustedIssuerCertificates>`, which means the issuer certificate is per se not trusted, but needed to complete the chain validation.
A certificate chain is trusted if *one* certificate of the chain is found in the `<TrustedPeerCertificates>` store.
Hope this makes sense..

elliot-gawthrop, 2022-10-10:
Thanks for the clarification, makes sense to me now. Happy for this to be closed.

closed 2022-10-11 by mregen