OPCFoundation/UA-.NETStandard #1945: How to send more than one app certificate in client ?

JsHuang, 2022-10-09:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

No document

### Expected Behavior

Is there any example of sending more than one application certificate while connecting to a opc ua server?
This is not the normal useage, but the OPC UA Reference support this kind of behavior.

mregen, 2022-10-26:
@JsHuang the only option is to send the whole cert chain instead of just the leaf cert or self signed cert. There is a configuration flag:
```xml
<SendCertificateChain>true</SendCertificateChain>
```
As you can see in the [OpenSecureChannelRequest](https://reference.opcfoundation.org/Core/Part4/v104/5.5.2/) only a single blob for the client application certificate exists.

mregen, 2023-01-07:
Please reopen if not answered...

closed 2023-01-07 by mregen