OPCFoundation/UA-.NETStandard #1948: HasApplicationSecureAdminAccess API Is Too Restrictive

elliot-gawthrop, 2022-10-10:
### Type of issue

- [X] Bug
- [X] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

The methods within the `ConfigurationNodeManager` cannot be run unless the `HasApplicationSecureAdminAccess` method can be executed without exceptions.

This method throws an exception if `ISystemContext.UserIdentity` is not a `SystemConfigurationIdentity`.

This API design is difficult to use. The server may its own implementation of `IUserIdentity`, and it may not be possible to derive from a new base class. The `HasApplicationSecureAdminAccess` is not virtual, so this behaviour cannot be changed.

romanett, 2024-01-15:
@elliot-gawthrop @mrsuciu with #2444 this issue  is fixed

romanett, 2024-04-04:
will solve with new Pull Request

PR #2578 (2024-04-11): Remove unnecessary casts to RoleBasedIdentity when checking for WellKnownRoles

## Proposed changes

Do not cast to RoleBasedIdentity when only IUserIdentity is needed because only WellKnownRoles are cheked.
This allows users to use own implementations of IUserIdentity

## Related Issues

- Fixes #1948

## Types of changes

What types of changes does your code introduce?

- [ ] Bugfix (non-breaking change which fixes an issue)
- [x] Enhancement (non-breaking change which adds functionality)
- [ ] Test enhancement (non-breaking change to increase test coverage)
- [ ] Breaking change (fix or fea [... cut]
closed 2024-04-11 by mregen