OPCFoundation/UA-.NETStandard #1951: AddUnsecurePolicyNone adds all SignAndEncrypt policies

elliot-gawthrop, 2022-10-12:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

The documentation for `IApplicationConfigurationBuilderServerPolicies.AddUnsecurePolicyNone` states that it adds the unsecure security policy type none to the server configuration.

Running this method will actually add the None security policy, and all SignAndEncrypt policies.

This is annoying because when attempting to add all SignAndEncrypt policies `RsaUtils.IsSupportingRSAPssSign` will be evaluated, which will then cause an exception to be thrown.

There is no workaround for this, because calling `AddPolicy(MessageSecurityMode.None, SecurityPolicies.None)` will throw an exception telling the user to use `AddUnsecurePolicyNone` instead! This seems like a pointless restriction.

### Expected Behavior

I expected that `IApplicationConfigurationBuilderServerPolicies.AddUnsecurePolicyNone` would only add the None policy, as per documentation.

I also expected that `AddPolicy(MessageSecurityMode.None, SecurityPolicies.None)` would allow me to workaround the above bug.

mregen, 2022-10-26:
Hi @elliot-gawthrop , in order to ensure a user has 'secure by design' in mind the API has been designed to require some extra effort to enable security None, so AddPolicy(None,None) is not supported for this reason. 
By default all secure policies are added which are not deprecated. Others need to be added manually, including None.
I think the only way to support your use case were by adding a parameter to AddUnsecurePolicyNone(bool singlePolicy) or so.

mregen, 2022-12-11:
changed `AddUnsecurePolicyNone` to only add that policy.

PR #2019 (2022-12-12): Add Discovery Url null check in PatchEndpointUrl
## Proposed changes

- Fix a null reference exception if an invalid discovery Url is returned by the server which cannot be parsed as Uri.
- Fix AddUnsecurePolicyNone to only add that policy.
- Fix a warning due to HttpHandler MaxConnectionsPerServer property

## Related Issues

- Fixes #2004 
- Fixes #1951 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-br [... cut]
closed 2022-12-12 by mregen