OPCFoundation/UA-.NETStandard #1958: The communication is abnormal due to high CPU usage

niubilityxixi, 2022-10-22:
### Type of issue

- [ ] Bug
- [X] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

In the same domain, there are threads with high CPU usage, resulting in slow event notifications and even connection failures.

### Expected Behavior

Reduces the impact of high CPU usage on connections.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

mregen, 2022-10-25:
@niubilityxixi please elaborate on the issue? There is otherwise not enough information to act.

niubilityxixi, 2022-10-25:
> @niubilityxixi please elaborate on the issue? There is otherwise not enough information to act.

The exception has been solved. Thank you for your reply.

niubilityxixi, 2022-10-25:
@mregen The exception has been solved. Thank you for your reply.

closed 2022-10-25 by mregen