OPCFoundation/UA-.NETStandard #1971: <title>

MarkusSchaffner, 2022-11-02:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

Scoped literal IPv6 addresses are not supported

### Expected Behavior

Can use IPv6 adresses like that: opc.tcp://[fe80:0:0:0:280:deff:fa02:c63e**%eth0**]:4840

### Environment

```markdown
- OS: Linux
- Environment: Ubuntu
- Runtime: .Net 5.0
- Nuget Version: 1.4.370.12
- Component:
- Server:
- Client:
```

### Anything else?

If using a scoped literal IPv6 address, the attachment to IPv6 address (e.g. "%eth0") will be deleted  and the device cannot be connected (in Linux). In Windows there is no issue like that.

Method: _Create(Uri discoveryUrl, EndpointConfiguration endpointConfiguration, IServiceMessageContext messageContext,      X509Certificate2 clientCertificate = null)_
Class: _DiscoveryChannel_ 
File:  _./UA-.NETStandard/Stack/Opc.Ua.Core/Stack/Client/DiscoveryClient.cs)._

by changing:
_endpoint.EndpointUrl = discoveryUrl.ToString();_ (in line 319)

to:
**endpoint.EndpointUrl = discoveryUrl.OriginalString;**

the issue will be fixed.

k1mur4jj, 2022-11-03:
I have exactly the same problem. Seems to be that the url.ToString() method will filter out the interface.

mregen, 2022-11-04:
@k1mur4jj and @MarkusSchaffner , is this the only place where this fix is needed? There is a second location with the same conversion for discoveryUrl. Have you checked that the selected endpoint connection can be done with the fix, not just the discovery?

MarkusSchaffner, 2022-11-04:
I can only speek aout the place i wrote. Because it worked for us, after changing that line.
Maybe it has to be changes in the other location, as well, if using the client in an other way like we do.

I think it make sense to change the "tostring()" call in all "create" methods (line 378, 347 in the same file)

If it have to be changed in other files (ConfigurationEndpoints, EndpointDescription, etc.) i cannot say.

But it sems to be not harmful to change it globaly.

PR #1976 (2022-11-16): On discovery preserve IPv6 scopeid for local link addresses
## Proposed changes

Using the uri.OriginalString instead of the 'ToString()' preserves a scope id of a IPv6 local link address.

## Related Issues

- Fixes #1971 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds functionality)
- [ ] Test enhancement (non-breaking change to increase test coverage)
- [ ] Breaking change (f [... cut]
closed 2022-11-16 by mregen