OPCFoundation/UA-.NETStandard #1975: Persist session across client reboots

davidguidali, 2022-11-04:
Our company runs an OPC UA client inside a docker container. This container is being updated frequently, causing it to restart. 
Since the OPC UA client makes use of _Alarms & Conditions_ capabilities, we have to make sure that we don't miss any data during this update process. 

We can not use the _TransferSubscription_ service, because **the OPC UA server we're connecting to does not support this service**. 
The server we use is the one of Siemens on the S7-1500 PLC, which is widely used in the industry. We already got in contact with Siemens and they told us that _TransferSubscription_ will not be supported, but we could increase the session timeout and re-connect to an existing session.

Unfortunately, there seems to be no way using the existing .NET Standard Client SDK to persist a session across client reboots, since the authentication token is stored in a protected property and can only be set during the creation of a new session. So my questions are:

1. Is there a recommended way to store session information and persist a session across client reboots using the UA .NET Standard SDK?
2. If not, is there any recommended work-around to support our scenario?
3. If there is no way to handle the scenario via the existing UA .NET Standard SDK, are there any plans to implement this functionality in the near future?

Thanks in advance

mregen, 2022-11-24:
Hi @davidguidali, we need to investigate what is needed to implement the support for 3.) and if it can be done in a secure manner. I know some people in my company did a prototype, but I need to dig up what was required. For 1) and 2) I have currently no answer.

davidguidali, 2023-01-27:
Are there already any updates on this issue?
We tried with different vendors which are widely used in the industry like Siemens and B&R, none of them is actually supporting the _TransferSubscription_ service, but they are able to persist sessions on the server and would support re-activating a session. Since we make use of A&C we require a persistent connection to the server to not lose any events. 
We refer to the OPC Standard 6.7 "Re-establishing connections" where it states:

_When a Client loses the connection to the Server, the goal is to reconnect without losing information. To do this the Client shall re-establish the connection by creating a new SecureChannel and activating the Session with the Service ActivateSession. This assigns the new SecureChannel to the existing Session and allows the Client to reuse the Session and Subscriptions in the Server. To re-establish the SecureChannel and activate the Session, the Client shall use the same security policy, application instance certificate and the same user credential used to create the original SecureChannel. This will result in the Client receiving data and event Notifications without losing information provided the queues in the MonitoredItems do not overflow._

_The Client shall only create a new Session if ActivateSession fails._

Due to the current implementation of the SDK, we're not able to implement persistence according to the standard.

See: https://reference.opcfoundation.org/Core/Part4/v104/docs/6.7

Thanks for your help.

PR #2244 (2023-08-03): Support for session reconnect from distributed clients
## Proposed changes

- Support Client.Session API to save and restore session configuration information as xml or in a serializable data structure.
- API tests
- Fixes #1975 
- Fixes #2249 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [ ] Bugfix (non-breaking change which fixes an issue)
- [x] Enhancement (non-breaking change which adds functionality)
- [x] Test enhancement (non-breaking change to increase test coverage)
- [ [... cut]
closed 2023-08-03 by mregen