OPCFoundation/UA-.NETStandard #1978: How Can I Scan OPC Ua Servers in my local machine ?

fatihkutuk, 2022-11-07:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

I want to scan opc servers on my local machine and find them endpoint urls. How can i do this ?

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

MD-V, 2022-11-07:
Hello @fatihkutuk ,

you can use the OPC UA Local Discovery Service (https://github.com/OPCFoundation/UA-LDS).

Servers (that have the option enabled) register on the LDS. An OPC UA client can then discover the registered servers.

fatihkutuk, 2022-11-07:
Okey thenk u @MD-V . I checked it but didnt understand, is there any lib in c# ?

MD-V, 2022-11-07:
@fatihkutuk Check https://github.com/OPCFoundation/UA-.NETStandard/blob/ea8ceb6d5951484c74f35ced20495454dddd7434/Libraries/Opc.Ua.Client/CoreClientUtils.cs#L52

fatihkutuk, 2022-11-07:
Oket @MD-V it works. Thank you very mutch.

closed 2022-11-07 by fatihkutuk