OPCFoundation/UA-.NETStandard #1993: Server.ServerDiagnostics.SubscriptionDiagnosticsArray is missing in v1.4.370.12

pc-avatar-7076, 2022-11-14:
The SubscriptionDiagnosticsArray seems to have gone missing in stack version 1.4.370.12. It is present in stack version 1.4.370.1. 

To reproduce:
1. Clone the repo and `git checkout 1.4.370.12`
2. In Visual Studio 2022 (or 2019) open the UA Reference solution and set the ConsoleReferenceServer as the startup project (you can use global.json to define the appropriate .NET target)
3. Build and execute the ConsoleReferenceServer
4. Connect a UA Client (i.e. UA Expert) and connect to the reference server endpoint (i.e. `opc.tcp://<host>:62541/Quickstarts/ReferenceServer`) using no security and Anonymous authentication
5. Browse `Root.Objects.Server.ServerDiagnostics` and observe that the SubscriptionDiagnosticsArray is missing

Notes:
- .NET 6 and NET Core 3.1 targets both reproduced this
- The SubscriptionDiagnosticsArray node reappears after `git checkout 1.4.370.1`
- Diagnostics are enabled in the Quickstarts.ReferenceServer.Config.xml - `<DiagnosticsEnabled>true</DiagnosticsEnabled>`
- `Opc.Ua.Server.DiagnosticsNodeManager.CreateSubscriptionDiagnostics` seems to complete successfully. The 'diagnostic enabled' flag seems to be respected here. 

UA Expert capture:
[image]

mregen, 2022-12-11:
@mrsuciu please check if the [PR](https://github.com/OPCFoundation/UA-.NETStandard/commit/313aa2a2499d8690cf719a67176e131517bb8b78) causes this issue.

romanett, 2024-08-16:
[image]
Diagnostics are available for the own session through the session diagnostiacs summary. An Admin uses the ServerDiagnostics to view the diagnostics using the DiagnosticsArray

closed 2024-08-16 by romanett