OPCFoundation/UA-.NETStandard #2002: Why is WriterGroupId in MetaDataMessage always 0?

Iveroxy, 2022-11-22:
When sending DataSets cyclicly via UADP, the subscriber receives a MetaDataMessage and the MetaDataReceived-Event of the currently active Application gets triggered. You can receive the DataSetWriterId from the MetaDataMessage correctly, but the field for WriterGroupId is always 0. Why? Pictures show JSON serialized MetaDataMessage-Objekt. Is there a way you can send the WriterGroupId of the described MetaDataMessage automatically? 

Thanks
[image]
[image]

mrsuciu, 2022-12-11:
@Iveroxy 

Sorry for the late answer.
The PubSub implementation is in a beta release. The WriterGroup is not encoded currently in the MetaDataMessage.

In order to be able to do it, one must currently change the source code (existing methods) to something like :

```
public UadpNetworkMessage(WriterGroupDataType writerGroupConfiguration, DataSetMetaDataType metadata)
            : base(writerGroupConfiguration, metadata)
        {
            UADPVersion = kUadpVersion;
            WriterGroupId = writerGroupConfiguration.WriterGroupId; //  <== Added to support WriterGroupId 
            DataSetClassId = Guid.Empty;
            Timestamp = DateTime.UtcNow;

            m_uadpNetworkMessageType = UADPNetworkMessageType.DiscoveryResponse;
            m_discoveryType = UADPNetworkMessageDiscoveryType.DataSetMetaData;

            SetFlagsDiscoveryResponseMetaData();
        }
```
and 

```
private void SetFlagsDiscoveryResponseMetaData()
        {
            /* DiscoveryResponse:
             * UADPFlags bits 5 and 6 shall be false, bits 4 and 7 shall be true
             * ExtendedFlags1 bits 3, 5 and 6 shall be false, bit 7 shall be true (erata 9):Bit 4 of ExtendedFlags1 shall be true
             * ExtendedFlags2 bit 1 shall be false and the NetworkMessage type shall be discovery response
             * */
            UADPFlags = UADPFlagsEncodingMask.PublisherId | UADPFlagsEncodingMask.ExtendedFlags1;
            ExtendedFlags1 = ExtendedFlags1EncodingMask.Security | ExtendedFlags1EncodingMask.ExtendedFlags2;
            ExtendedFlags2 = ExtendedFlags2EncodingMask.NetworkMessageWithDiscoveryResponse;

            // enable encoding of PublisherId in message header 
            NetworkMessageContentMask = UadpNetworkMessageContentMask.PublisherId;
            // enable encoding of WriterGroupId in group header
            NetworkMessageContentMask |= UadpNetworkMessageContentMask.WriterGroupId; //  <== Added to support WriterGroupId 
        }
```

I did not run the code but the changes show the idea.

monikavar, 2023-03-23:
@Iveroxy, hope the above answer helps. Please feel free to re-open if you still need assistance. Thanks!

closed 2023-03-23 by monikavar