OPCFoundation/UA-.NETStandard #2007: Get Trusted Certificates from Server without GDS

yagizhanNY, 2022-11-29:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

I am trying to receive certificate trust list from server without GDS. Our app is MAUI based so we don't want to use GDS for certificate management.

I did some search about it but I couldn't find how can I handle that.

### Expected Behavior

I want to manage server side certificates without GDS.

### Environment

```markdown
- OS:Windows 10
- Environment:Visual Studio 2022
- Runtime: .NET6
- Nuget Version:
- Component:Opc.Ua.Client
- Server: Opc UA device
- Client:
```

### Anything else?

Is there any sample or documentation for this case?

mregen, 2022-12-01:
Hi @yagizhanNY, as a client you can use the GDS client library to connect with sysadmin privilege and read the server Trustlist/Issuerlist.

yagizhanNY, 2022-12-02:
Hi @mregen, thanks for comment. I tried to handle this case with GDS client library, it has worked!

Now I am able to manage certificates. Thanks.

closed 2022-12-02 by yagizhanNY