OPCFoundation/UA-.NETStandard #2010: JSON encoded DataValue timestamps should omit picoseconds

mregen, 2022-12-05:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

Source/Servertimestamp has the full resolution
```json
{
  "Timestamp": "2022-11-30T13:12:04.6528136Z",
  "Payload": {
    "dtmi:com:example:Thermostat;1:temperature": {
      "SourceTimestamp": "2022-11-30T13:12:04.6201185Z",
      "Value": 10455
    }
  },
  "DataSetWriterName": "thermostat-sample",
  "SequenceNumber": 2968
}
```

### Expected Behavior

Source/Servertimestamp should be rounded to milliseconds
```json
{
  "Timestamp": "2022-11-30T13:12:04.6528136Z",
  "Payload": {
    "dtmi:com:example:Thermostat;1:temperature": {
      "SourceTimestamp": "2022-11-30T13:12:04.620Z",
      "Value": 10455
    }
  },
  "DataSetWriterName": "thermostat-sample",
  "SequenceNumber": 2968
}
```

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

ZbynekZ, 2022-12-05:
Hello.

Why do you think the timestamps should be rounded to milliseconds? Is there a justification for it in the specification? 

The total precision of the timestamps can go, in general case, to tens of picoseconds. The DateTime structure in .NET is not so precise, it can go to 100 nanoseconds, but that is still 4 orders of magnitude more precise than milliseconds.

I understand that many applications won't use timestamps this precise, and the additional digits seem to be an overkill. But unless the specification mandates otherwise, I think that the precision in the JSON encoder should either be the maximum the system allows, or it should be configurable by the developer - but not artificially and forcefully "crippled down".

mregen, 2022-12-05:
Hi @ZbynekZ, good catch, it is supposed to be rounded to µs. So on .NET only the last digit is lost. Sound ok?

mregen, 2022-12-05:
_The Picoseconds fields store the difference between a high-resolution timestamp with a resolution of 10 picoseconds and the Timestamp field value which only has a 100 ns resolution. The Picoseconds fields shall contain values less than 10 000. The decoder shall treat values greater than or equal to 10 000 as the value ‘9999’._

closed 2022-12-05 by mregen