OPCFoundation/UA-.NETStandard #2012: <title>OPC UA Auto ID / RFID Reader Manage Antennas

enesceb, 2022-12-06:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

Can we manage the antennas using OPC UA
Ex: Antenna Open/Clove or If a data was read, which antenna was read?
How can I detect which antenna read a tag in OPC UA

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client: UA EXPERT
```

mregen, 2022-12-11:
@enesceb yes, please check the companion spec on opcfoundation.org how to code a client against AutoID devices.
As a hint, you can at least use the latest https://github.com/OPCFoundation/UA-ModelCompiler docker with Nodeset2 support to create the AutoID namespace constants from [here](https://github.com/OPCFoundation/UA-Nodeset/blob/latest/AutoID/Opc.Ua.AutoID.NodeSet2.xml) for the client.

monikavar, 2023-03-23:
@enesceb, please re-open if the issue still persists.

closed 2023-03-23 by monikavar