OPCFoundation/UA-.NETStandard #2014: GDS certificate push - built-in CA in GDS server doesn't preserve X500DistinguishedName-encoding from signing request.

decuse, 2022-12-07:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

Using the provided example GDS paired w, GDS client for certificate push using signing requests will create server certificates without preserving the ASN.1 encoding of the subject-name part of the CSR, which will lead to the server rejecting the certificate (unless you are lucky) with error: "Subject Name of new certificate doesn't match the application". 

On closer inspection, the GDS CA implementation seems to handle the signing request in [this function](https://github.com/OPCFoundation/UA-.NETStandard/blob/67a794e436e6910a902fb8ee1495fbed96659642/Libraries/Opc.Ua.Gds.Server.Common/CertificateGroup.cs#L216). If you look at line 263, you can notice that the subject is transformed into a string: 

`..., info.Subject.ToString(true, (IDictionary)Org.BouncyCastle.Asn1.X509.X509Name.DefaultSymbols), ...`

This means that any ASN.1 encoding in the subject name is not handled by the CertificateFactory, i.e., the encoding of the elements will to some extent be random.

Checking the code in ConfigurationNodeManager, which receives the new certificate in OPC UA server – this is as far as I understand done in [this function](https://github.com/OPCFoundation/UA-.NETStandard/blob/9344a2e9e62473c04b264cfa977a266e2759ac1a/Libraries/Opc.Ua.Server/Configuration/ConfigurationNodeManager.cs#L369). And on line 424: 
`if (!X509Utils.CompareDistinguishedName(certificateGroup.ApplicationCertificate.Certificate.SubjectName, newCert.SubjectName)) 
{
throw new ServiceResultException(StatusCodes.BadSecurityChecksFailed, "Subject Name of new certificate doesn't match the application.");
}`

Checking the CSR and new certificate respectively, the ASN.1 encodings for the SubjectName differs, but only on string encodings, e.g., CN is encoded is UTF8String in the CSR but as PrintableString in the new cert.

### Expected Behavior

I would expect the built-in CA-function of the GDS to be interoperable with the default ConfigurationNodeManager implementation. 

There are as far as I can see two different ways of fixing this:
1. Making the check in the ConfigurationNodeManager in the server less restrictive, e.g., by allowing the string-encodings to be different as long as the content is equal.
2. Using and preserving the ASN.1 encoding when creating and signing the certificate in the GDS.

### Steps To Reproduce

1. Run the GDS and GDS-client from the UA Global Discovery Server
2. Connect GDS-client to GDS (login w. appadmin/demo), accept certificate of GDS, etc.
3. Run any standard server, in which you can elevate the user to sysadmin.
4. in GDS client select Registration - Registration Type: Server - Push Management
5. Click Pick Server and type the address of your OPC UA server, login using the sysadmin credentials
6. Server State - press green arrow
7. Certificate - press "Request New"
8. If doing everything correct, text saying "Certificate request in process" (or something similar) is displayed in UI. After some time the error message: "Subject Name of new certificate doesn't match the application." is displayed.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

decuse, 2022-12-08:
As a comment - I can make it (almost) work if I use the .net-core versions on both GDS and the OPC UA server - it seems as then the ASN.1 encodings end up being the same.

mregen, 2022-12-11:
Hi @decuse, thanks for the report, the solution is to preserve the ASN.1 encoding, it is mandated by some CA browser forum rules that Issuer/Subject of certs are binary identical.

closed 2022-12-12 by mregen

decuse, 2022-12-13:
Thank you @mregen for the fast handling of this issue!

PR #2018 (2022-12-12): CSR doesn't preserve ASN.1 encoding of the Subject for GDS push
## Proposed changes

- Preserve the ASN.1 encoding of a CSR Subject in the issued certificate.
- However, a certificate issuer is allowed to change the Subject. Make the check on GDS Push only check the Subject string, to ensure the app config can still find the pushed cert after restart.
- found an issue during debugging the session disconnect may run into unnecessary exceptions as `ChannelShutdown` may be  called twice.

## Related Issues

- Fixes #2014 

## Types of changes

What types of changes does your code introduce?
_Put  [... cut]