OPCFoundation/UA-.NETStandard #2023: Plans for release of 1.5 version

ZbynekZ, 2022-12-17:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

Currently the 1.5 branch is in pre-release, on NuGet it is 1.5.371.3-beta.

### Expected Behavior

Are there plans to release it, and if so, roughly when? I specifically need encoding/decoding datatypes like PubSubConfiguration2DataType which are not in 1.4 but are already in 1.5.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

mregen, 2023-01-11:
Hi @Zbynek, we need to discuss, please stay tuned...

ZbynekZ, 2023-04-01:
@mregen, for me, simply adding 1.05 classes to the 1.04 code will work perfectly. I am doing client development and need the types. And it cannot currently be accomplished easily by "borrowing" them from 1.05, because the NuGet package already contains the 1.04 types. I can add types one-by-one (just those that I need), but it is a lot of work and even that has problems (encoder interface has changed). 

Server development might be different (1.05 nodes will appear in the address space?), but I am not concerned with it.

EthanChangAED, 2024-03-20:
The .net stack officially moved to 1.05 from January release. Feel free to file another issue if there is any feature you need missing.

PR #2466 (2024-01-18): Cherry pick hotfix and tag as 1.5.372 release.
## Proposed changes

The 1.5 nodeset has been updated and tested. flag as release.
This release is on parity with the 1.4.372.107 release, but contains the latest Nodeset as of the 1.05.03 spec release.
Includes hotfix for 1.4.372.107.

## Related Issues

- Fixes #2023 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [x] Bugfix (non-breaking change which fixes an issue)
- [x] Enhancement (non-breaking change which adds functionality)
- [... cut]
closed 2024-03-20 by EthanChangAED