OPCFoundation/UA-.NETStandard #2027: Value Attribute Reads on Server_ServerStatus_BuildInfo components return defaults

jleger-ptc, 2022-12-21:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

Starting in release 1.4.371, I am no longer able to perform Value attribute reads on the Root -> Objects -> Server -> ServerStatus -> BuildInfo child objects/components. They now return good quality but with 'Null'/default values (depending on the data type).

Reverting back to the 1.4.370 release returns the data as expected

UA Expert Result for **BuildInfo**

[image]

UA Expert Result for **ProductName**

[image]

### Expected Behavior

The sub-component values should match those reported in the BuildInfo complex object

### Steps To Reproduce

1. Check out the **release/1.4.371** tag
2. Open the "UA Reference" solution
3. Build and run the **ConsoleReferenceServer** project
4. Connect UA Expert to the Reference Server
5. Browse to the Root -> Objects -> Server -> ServerStatus -> BuildInfo object in the address space
6. Click on the BuildInfo object to perform an Attribute read
7. Browse into the BuildInfo object
8. Click on the ProductName object to perform an Attribute read

### Environment

```markdown
- OS: Windows 10
- Environment: Visual Studio 2022 17.2.6
- Runtime: .NET Core 3.1
- Nuget Version: 1.4.371.49
- Component: Reference Server
- Server: Reference Server
- Client: UA Expert
```

mregen, 2022-12-21:
good catch, thanks!

mregen, 2022-12-21:
works in 1.4.371.35-preview docker ref server, broken afterwards. Check https://github.com/OPCFoundation/UA-.NETStandard/pull/1984

mregen, 2022-12-22:
Modelcompiler removed code to handle properties of complex types because it didn't work correctly under all circumstances: 
[Remove unused types from NodeSet + removed broken nested Variable sup… · OPCFoundation/UA-.NETStandard@faebffc (github.com)](https://github.com/OPCFoundation/UA-.NETStandard/commit/faebffc1cf2525cd17ac577bd5b8c675b65471b6)
Needs to be implemented manually or help wanted to fix Modelcompiler.

mregen, 2023-07-03:
1.4.371.96 with the latest nodeset update exposes again the issue
fix planned for 1.4.372 -> #2212

EthanChangAED, 2024-03-21:
Closing this issue as fix has been released.

PR #2064 (2023-02-20): Default SendCertficateChain=true and temp fix for BuildInfo
## Proposed changes

- After discussion default SendCertChain to true
- Bring back part of the code which was removed by Modelcompiler to fix the BuildStatus variable
- Fix the GDS Server Push, which does not fully update the server descriptions with a chain, if sendcertchain is selected

## Related Issues

- Fixes #2020 
- Fixes #2027 
- Fixes #2079 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [x] Bugfix (non-breaking [... cut]
closed 2024-03-21 by EthanChangAED