OPCFoundation/UA-.NETStandard #2032: OPCUA client should check the application URI of the server certifiate

GregoireG-C, 2022-12-27:
### Type of issue

- [X] Bug
- [X] Enhancement
- [X] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

The OPCUA client verify the following points concerning the server certificate:
- it is trusted
- its certificate chain is valid
- its certificate chain is completely known
- its clr is valid and none of the certificate chain is revoked
- its time validity is not expired or not 'not yet valid'
-The DNS match the server machine 
...

But the client does not check the ApplicationUri of the server presented in its application certificate.

### Expected Behavior

Th client should verify the applicationUri of the server certificate.

### Steps To Reproduce

1.Compile sample client and standard UA server
2. Connect using sign or sign&encrypt to the server ; the server using a certificate presenting an applicationUri null or different form the one present into its configuration file
3. The client accept the connection.

### Environment

```markdown
- OS:Windows 10
- Environment: Visual Studio 2019
- Runtime:NET 4.6.2
- Server: Reference server
- Client: Reference client
```

GregoireG-C, 2022-12-27:
According to the standard part 4 - Service Release:
§5.4.1 Discovery Service Set:

> [...]
> A Client shall be careful when using the information returned from a DiscoveryEndpoint since it
> has no security. A Client does this by comparing the information returned from the
> DiscoveryEndpoint to the information returned in the CreateSession response. A Client shall verify
> that:
> a) The ApplicationUri specified in the Server Certificate is the same as the ApplicationUri
> provided in the EndpointDescription.
> [...]

=> This suggest that the client must verify that the applicationUri presented in the endpoint descriptions and in the server certificate in the GetEndpointResponse match.

GregoireG-C, 2022-12-28:
I suggest the following modification:
Opc.Ua.Client.Session
public method open:

```
if (checkDomain)
{
    m_configuration.CertificateValidator.Validate(serverCertificateChain, m_endpoint);
    //Modification - Begin
    // verify if applicationUri from ApplicationDescription matches the applicationUri in the client certificate.
    string certificateApplicationUri = X509Utils.GetApplicationUriFromCertificate(serverCertificate);
    if (!String.IsNullOrEmpty(certificateApplicationUri) &&
        !String.IsNullOrEmpty(m_endpoint.Description.Server.ApplicationUri))
    {

        if (certificateApplicationUri != m_endpoint.Description.Server.ApplicationUri)
            throw new Exception(
                string.Format("Server certificate is refused. The applicationUri does not match:" +
                "-Certificate applicationUri='{0}', -EndpointDescription applicationUri='{1}'",
                certificateApplicationUri, m_endpoint.Description.Server.ApplicationUri));
        else
            throw new Exception(
                string.Format("Server certificate is refused. Server certificate is refused, the applicationUri " +
                "in the certificate or in the endpoint description is null."));
    }
    //Modification - End
}
else
{
    m_configuration.CertificateValidator.Validate(serverCertificateChain);
}

```

mregen, 2024-04-05:
The application Uri should always be checked against the information in the endpoint.

mregen, 2024-04-09:
Hi @GregoireG-C, is this topic no more an issue? We were planning to include your fix in the next release, please share if it is not necessary.

GregoireG-C, 2024-04-09:
It's still an issue. I closed it because you have added this one to a milestone. So I have not any reason to follow it. I will let you close it then.

Best regards

ykarpeev, 2024-06-07:
Hello, is there a recommended way to bypass this functionality other than modifying the code on my end as I am doing now?

My use is case is that I often develop against opc-ua servers that are on internal LANs - so I setup a computer on that LAN and then use netsh to port proxy over. It looks like this causes the certificateApplicationUri to not match the applicationUri.

I think it is because GetApplicationUriFromCertificate return's the URI of the PC that I am using as the proxy. 

I also want to be able to connect to servers even if they have invalid certificates in the future.

romanett, 2024-06-07:
Hi, my recommendation would be to use the "Security: None"  endpoint as all the other endpoints need to follow the spec

ykarpeev, 2024-06-07:
Thank you. That works great here. Below is what I added.

```
            EndpointDescription endpointDescription = CoreClientUtils.SelectEndpoint(this.configuration, serverUrl, true);

            endpointDescription.SecurityPolicyUri = SecurityPolicies.None;
            endpointDescription.SecurityMode = MessageSecurityMode.None;
```

mregen, 2024-08-26:
at this time the check has been removed to work around IOP issues. WIP #2731

mregen, 2025-02-26:
An update on Mantis was posted by the WG: [Mantis](https%3A%2F%2Fmantis.opcfoundation.org%2Fview.php%3Fid%3D9795&data=05%7C02%7CMartin.Regen%40microsoft.com%7C5a1c060702a0496d990408dd55bc758e%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C638760990147093996%7CUnknown%7CTWFpbGZsb3d8eyJFbXB0eU1hcGkiOnRydWUsIlYiOiIwLjAuMDAwMCIsIlAiOiJXaW4zMiIsIkFOIjoiTWFpbCIsIldUIjoyfQ%3D%3D%7C0%7C%7C%7C&sdata=a5tyKHrd8rg3a%2B19O4gKhTjavPs2Wf9uIyqb8d7A8Ao%3D&reserved=0)

PR #2583 (2024-05-02): check Application URI of the server Certificate on OpenSecureChannel
## Proposed changes

Implement OpenSecureChannel in compliance with 
[Spec Part 5.4.1](https://reference.opcfoundation.org/Core/Part4/v105/docs/5.4.1)

The ApplicationUri specified in the Server Certificate is the same as the ApplicationUri provided in the EndpointDescription.

## Related Issues

- Fixes #2032

## Types of changes

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds functionality)
- [ ] Test enhancement (non-breaking change to increase test coverage)
- [ ] [... cut]