OPCFoundation/UA-.NETStandard #2037: Reg: AMQP Pub Sub Connection Implementation

BalajiGuptaK, 2023-01-05:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

Though I am able to see AMQP in TransportProtocol enum, there is no class implementation for this protocol like MqttPubSubConnection class.

Is this not yet implemented feature?
If it is already implemented, then which broker will be the best option for this AMQP.

Please clarify.

### Environment

```markdown
- OS:Windows 10
- Environment:VS 2019
- Runtime: netstandard2.0
- Nuget Version:1.4.369.30
- Component:Opc.Ua.PubSub
- PubSub: UA PubSub Preview
```

mregen, 2023-01-05:
Hi @BalajiGuptaK, currently there is more traction around the MQTT implementation and thats the protocol I would recommend.

mregen, 2023-02-28:
please reopen if question notanswered

closed 2023-02-28 by mregen