OPCFoundation/UA-.NETStandard #2046: ASP.Net core dependancy since 1.4.368.58

Jordu29, 2023-01-10:
Hi, 
I have a .NET Core 6 project using the OPCFoundation.NetStandard.Opc.Ua package. The 1.4.368.58 version introduces an ASP.NET Core dependancy as shown below 
[image]

No dependancies occur when I downgrade every project to 1.4.367.100 : 
[image]

mregen, 2023-01-11:
Hi @JordanHoareau, the original UA package has been split into subpackages for client and server to fine tune dependencies. Https has become an optional package. The problem with https support is the kestrel server where version 2.x is now end of life and the new way to distribute it is in asp.net. If you switch to the subpackages the dependencies will go away but you don't have the https support.

mregen, 2023-02-25:
please reopen if not answered

closed 2023-02-25 by mregen