OPCFoundation/UA-.NETStandard #2048: FastDataChangeCallback is not awaiting

TeaTime762, 2023-01-12:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

When SequentialPublishing in Subscription is enabled the callback is executed in a single-threaded manner according to the description.
But I see in the code that the invoking of FastDataChangeCallback is not awaited. This means the next callback can overwhelm the former callback. So it a "multi-threaded" behaviour.
Is my train of thought correct?

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

mregen, 2023-04-08:
Hi @TeaTime762, when the `SequentialPublishing` is enabled, the Tasks which execute are serialized by a semaphore. The FastDataChangeCallback is not an async call and doesn't need to be awaited. So no multi threading. Does this answer your question?

TeaTime762, 2023-04-17:
Yeah, thanks!

closed 2023-04-17 by TeaTime762