OPCFoundation/UA-.NETStandard #2059: Stack overflow when subscribing to server diagnostic nodes

jnsjll, 2023-01-24:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

With server diagnostics enabled, subscribing to server diagnostic nodes can lead to a stack overflow on the server side in certain situations.

### Expected Behavior

No stack overflow is experienced.

### Steps To Reproduce

- Add a Thread.Sleep to the DiagnosticsNodeManager DoScan method to simulate many sessions / subscriptions.
[image]
- Run the reference server 
- Connect with a client 
- Recursively subscribe to all hierarchically referenced variable / variable types (make sure to increase any max recursive child nodes setting accordingly) starting from the Root / Objects node
- Wait for a stack overflow

Note: the issue occured without the Thread.Sleep but it is somehow difficult to reproduce.

### Environment

```markdown
- OS: Windows 10
- Environment: 
- Runtime: .NET 4.6.2
- Nuget Version: 1.4.368.53
- Component: Opc.Ua.Server
- Server: Reference Server
- Client: UA Expert
```

### Anything else?

The occurence of the problem seems to depend both on the configured PublishResolution setting (calling `DiagnosticsNodeManager.ForceDiagnosticsScan` every x ms) and the duration it takes to update and report notifications for monitored diagnostics node in `DiagnosticsNodeManager.DoScan` and can lead to this recursion:

[image]

mregen, 2023-04-08:
Hi @jnsjll, your fix #2060 would fix the recursion issue, however it creates a new lock contention between the subscriptions and the diagnostic server lock. Could you provide a fix which just prevents the recursion of DoScan?

mregen, 2023-04-11:
#2060 merged

closed 2023-04-11 by mregen