OPCFoundation/UA-.NETStandard #2066: Remove the support for .NET Framework 4.6.2

mregen, 2023-01-27:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

Currently on .NET Framework 4.6.2 the Nuget packages can only support the RSA type certificates with the help of bouncy castle. 
.NET Framework 4.8. supports the CertificateRequest API needed to get rid of bouncy castle for certificate handling, in addition it supports ECC nist and brainpool on the platform. To simplify the future support .NET 4.6.2 is deprecated in the next breaking change release 1.4.372. Moving forward only .NET Framework 4.8 is supported.

### Environment

```markdown
- OS: Windows
- Environment:
- Runtime:.NET Framework 4.6.2
- Nuget Version:
- Component:
- Server:
- Client:
```

eg-mt, 2023-02-04:
Hi @mregen would this prevent a .NET 4.6.2 application that references a .NET standard 2.0 assembly using the UA-.NETStandard nuget?

mregen, 2023-02-25:
> Hi @mregen would this prevent a .NET 4.6.2 application that references a .NET standard 2.0 assembly using the UA-.NETStandard nuget?

Technically not, but netstandard2.0 is underlying the same restrictions for the use of bouncy castle and the ECC support, so support for the netstandard2.0 may also be deprecated.

closed 2023-05-31 by mregen