OPCFoundation/UA-.NETStandard #2073: session.Load() does not load MonitoredItems

andnig, 2023-02-06:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

When creating a Session and then loading subscriptions via ``session.Load("path_to_list_of_subscriptions")`` , only the subscriptions and DefaultItems - but not the MonitoredItems are loaded.

Up until at least version 1.4.365.48, the Load() - Method also correctly serialized the MonitoredItems which were defined in a subscription file.

## More reference

I create the session as follows

```C#
            var channel = SessionChannel.Create(
                applicationConfiguration,
                endpoint.Description,
                endpoint.Configuration,
                clientCertificate,
                messageContext);

                var newSession = new Session(channel, applicationConfiguration, endpoint, null)
                {
                    ReturnDiagnostics = DiagnosticsMasks.All,
                };
``` 

Then I load the subscriptions as follows

```C#
var subscriptions = newSession .Load("path_to_list_of_subscriptions.xml")
```

All the subscriptions and DefaultItems get loaded, but the MonitoredItems - List is empty.
Up until version 1.4.365.48 this worked fine.

### Example Subscriptions File

```xml
<?xml version="1.0" encoding="utf-8"?>
<ListOfSubscription xmlns:i="http://www.w3.org/2001/XMLSchema-instance" xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd">
  <Subscription>
    <DisplayName>Subscription 1</DisplayName>
    <PublishingInterval>1000</PublishingInterval>
    <KeepAliveCount>10</KeepAliveCount>
    <LifetimeCount>10</LifetimeCount>
    <MaxNotificationsPerPublish>1000</MaxNotificationsPerPublish>
    <PublishingEnabled>true</PublishingEnabled>
    <Priority>100</Priority>
    <TimestampsToReturn>Both_2</TimestampsToReturn>
    <MaxMessageCount>10</MaxMessageCount>
    <DefaultItem>
      <DisplayName>MonitoredItem</DisplayName>
      <StartNodeId i:nil="true" />
      <RelativePath i:nil="true" />
      <NodeClass>Variable_2</NodeClass>
      <AttributeId>13</AttributeId>
      <IndexRange i:nil="true" />
      <Encoding i:nil="true" />
      <MonitoringMode>Reporting_2</MonitoringMode>
      <SamplingInterval>-1</SamplingInterval>
      <Filter i:nil="true" />
      <QueueSize>0</QueueSize>
      <DiscardOldest>true</DiscardOldest>
    </DefaultItem>
    <MinLifetimeInterval>10000</MinLifetimeInterval>
    <MonitoredItems>
      <MonitoredItem>
        <DisplayName>Int32Value</DisplayName>
        <StartNodeId>
          <Identifier>ns=2;i=10849</Identifier>
        </StartNodeId>
        <RelativePath i:nil="true" />
        <NodeClass>Variable_2</NodeClass>
        <AttributeId>13</AttributeId>
        <IndexRange i:nil="true" />
        <Encoding i:nil="true" />
        <MonitoringMode>Reporting_2</MonitoringMode>
        <SamplingInterval>1000</SamplingInterval>
        <Filter i:nil="true" />
        <QueueSize>0</QueueSize>
        <DiscardOldest>true</DiscardOldest>
      </MonitoredItem>
    </MonitoredItems>
    <DisableMonitoredItemCache>false</DisableMonitoredItemCache>
  </Subscription>
</ListOfSubscription>
```

### Expected Behavior

The MonitoredItems as defined in the subscriptions file should be serialized and be part of the loaded subscription object.

### Steps To Reproduce

1. Create a Session
2. Use session.Load() to load a subscriptions.xml file

### Environment

```markdown
- OS: Windows 10
- Environment: Visual Studio 2019
- Runtime: .NET 6
- Nuget Version: 1.4.367.75 (and all following. With 1.4.365.48, the behavior was as expected)
- Component: Opc.Ua.Client
- Server:
- Client:
```

mregen, 2023-04-08:
Hi @andnig, thanks for the report, there are a few issues to fix here:
i) there are tests for loading of the saved subscriptions and it looks like the  monitored items are still deserialized
if you move `<MinLifetimeInterval>10000</MinLifetimeInterval>` after the `<MonitoredItems/>`.
There was the same order in the subscription class assigned to both MonitoredItems and MinLifeTimeinterval. 
However, after assigning each a seperate order the outcome was incompatible to previously saved subscriptions, as you have noticed because order was effectively reversed. 

ii) Another issue found was that the monitored items are not created on the server because the `serverId` is now saved with the monitored items to allow to transfer a subscription after a session reconnect. However, loading the serverId from file ends up in a regression that the monitored items are not recreated on a server in the normal use case, not trying to transfer a subscription.

A fix to support both cases were to add a flag to the `Load` function to indicate that a session transfer should be supported. Otherwise the serverId is reset after load.

closed 2023-04-11 by mregen

andnig, 2024-01-20:
Hey @mregen I apologize for coming back to this issue a year later... I had a workaround in place, but the mistakes of the past are hunting me :-D

With regards to the serialization problem number i) above: Is there a way how I can use Load() without needing to move <MinLifetimeInterval>10000</MinLifetimeInterval> after the <MonitoredItems/>? I have some configurations in the wild where I can't change the subscription config easily (also not programmatically, due to file permissions) - so it would be great if I somehow could get the MonitoredItems serialized, even when the MinLifetimeInterval is configured above them...

PR #2113 (2023-04-11): Session.Load doesn't create the monitored items
## Proposed changes

- Issue: The monitored items are not created on the server because the serverId is now saved with the monitored items to allow to transfer a subscription after a session reconnect. However, by loading the serverId from file there is a regression that the monitored items are not recreated on a server in the normal use case, when somebody is not trying to transfer a subscription.

- Fix: to support both use cases added a flag to the Load function to indicate that a session transfer should be supported. Otherwise the serverId is [... cut]