OPCFoundation/UA-.NETStandard #2086: Create Session Error : TypeInitialization_Type, Opc.Ua.EncodeableFactory

sensiki, 2023-02-28:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

ConsoleReferenceClient application uses NET7, which is incorrect under Linux, but correct under Windows.
.NET6 all right

error:
Create Session Error : TypeInitialization_Type, Opc.Ua.EncodeableFactory

### Environment

```markdown
- OS:linux
- Environment: vs2022
- Runtime:NET7.0
- Nuget Version:
- Component:
- Server:
- Client:
```

mregen, 2023-02-28:
@sensiki the samples are not using .NET7 yet, it was just tested if the libraries can be compiled. Could you elaborate on the issue you experience?

sensiki, 2023-03-01:
@mregen I have solved this problem and can now be used normally.The reason is trim. NET7 needs "<TrimMode>partial</TrimMode>"
Thank you.

closed 2023-03-01 by sensiki